rsta2/circle #168: USB PnP and fatfs add-on (fix included)

olfp, 2020-12-02:
Hi! I found that when you unplug a USB memory stick while a kernel is running, the PnP layer takes notice but the fatfs add-on does not. On the next access to the removed drive the kernel crashes hard.

After learning more, I found that registering a simple callback in adding/fatfs/diskio.cpp solved the problem. I replaced the disk_initialize function with this (adding the callback fn and registering it):
void disk_removed (
        CDevice *pDevice                        /* device removed */
,       void *pContext
)
{
  *((CDevice **)pContext) = (CDevice *)0;
}

DSTATUS disk_initialize (
        BYTE pdrv                               /* Physical drive nmuber to identify the drive */
)
{
        if (pdrv >= FF_VOLUMES)
        {
                return STA_NOINIT;
        }

        s_pVolume[pdrv] = CDeviceNameService::Get ()->GetDevice (s_pVolumeName[pdrv], TRUE);
        if (s_pVolume[pdrv] != 0)
        {
                s_pVolume[pdrv]->RegisterRemovedHandler (&disk_removed, &s_pVolume[pdrv]);
                return 0;
        }

        return STA_NOINIT;
}

Since the higher level fatfs functions check the disk status (which is invalidated by disk_removed) they now gracefully fail.

I hope this will be useful to others too.

Best regards
Olaf

rsta2, 2020-12-03:
Does it also crash, if FatFs is unmounted, before the USB flash drive is removed, with:

```
f_mount (0, "USB:", 0);
```

The problem with remove-able disks is, that they have to be unmounted to keep the file system OK. They shouldn't be surprise-removed.

olfp, 2020-12-03:
When the FatFs is unmounted all is fine. While disks should not be surprise-removed, in reality they are. My aim is that the system should not crash hard in such a case.

rsta2, 2020-12-03:
I see. Thanks for providing the patch!

rsta2, 2021-01-08:
Your fix is now included into Circle on the *develop* branch. Thanks!

closed 2021-01-08 by rsta2