rsta2/circle #169: How to send patches (ANSI color & control-cursor keys)

olfp, 2020-12-07:
Hi, I would like to share to patchsets, what's the preferred way? 
I have created support for ANSI 16 color support in the console and keyboard escape sequence support for ctrl-cursor keys.

Best regards
Olaf

rsta2, 2020-12-08:
What do you mean with "share"? If your patches should become part of Circle itself, you can send them to the Email address given at my main GitHub page. Of course, they have to work on all models and with all configurations then. You can also send a Pull Request on GitHub. Please branch off from the *develop* branch, because development is done here.

olfp, 2020-12-08:
HI! With "share" I mean make available to the other Circle users. I think they should become part of Circle. I prepare the patches and send them by Email. Thanks for the info.

closed 2020-12-08 by olfp