rsta2/circle #170: Compile Issue for Raspberry Pi 4

mitMathe, 2020-12-09:
Hi.

First of all, appreciate it for gorgeous source code.

For Raspberry Pi 1 and 2, the compile is well-done.

But, even though I modified the RASPPI ?= 4, I met some problem as below.

arm-none-eabi-ld: error: kernel7l.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/libgcc.a(_udivmoddi4.o) does not
arm-none-eabi-ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/libgcc.a(_udivmoddi4.o)
../../Rules.mk:154: recipe for target 'kernel7l.img' failed

Could you give some advices to me for solving this issue?

Best Wishes,
Yoo-Seung.

rsta2, 2020-12-09:
Thank you! I guess, you should use a newer toolchain to solve this problem. You can download a toolchain [here](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-a/downloads). Please use the version 9.2-2019.12 (at the bottom of the page), because the 10.2-2020.11 has not been tested with Circle yet.

mitMathe, 2020-12-11:
Thanks for prompt and kind reply.

I solved this issue after installing version 9.2 toolchain.

Best Wishes,
Yoo-Seung.

closed 2020-12-11 by mitMathe