rsta2/circle #172: 50MHz SD card clock speed

dwhinham, 2020-12-16:
Currently, `emmc.cpp` sets up the SD card interface with a fixed speed of 25MHz, and doesn't attempt to negotiate a faster rate of 50MHz either at 3.3V ("High Speed" mode) or 1.8V (SDR25 mode).

For Circle projects that do considerable amounts of file I/O, we can almost double the loading speed of larger files if we can switch to the higher speed mode. In [my project](/dwhinham/mt32-pi), users often have to wait for a large SoundFont file to load - some are only ~30MB, but others are >250MB, and so this has a big impact on loading times.

I've been measuring load times using a hack to change [this line](https://github.com/rsta2/circle/blob/42c0812385fe3b93c8f13b2410b4495c2d3fc385/addon/SDCard/emmc.cpp#L1703) to read `SwitchClockRate (base_clock, SD_CLOCK_HIGH);`, and it works very well, but it would be better if we can check the SD card's capabilities and conditionally switch to the faster speed for better compatibility.

Would you be open to a patch that tries to implement this? I'm not an SD card expert but I could try to read some specs and attempt this - unless you think you would be able to add this more easily than me! 🙂

I also noticed that this definition for 1.8V mode is commented out:
https://github.com/rsta2/circle/blob/42c0812385fe3b93c8f13b2410b4495c2d3fc385/addon/SDCard/emmc.cpp#L69

Was there a reason for this, and does it still apply to modern Raspberry Pis?

Thanks for your consideration!

rsta2, 2020-12-16:
Reliable SD card access is not as easy to achieve, as it might look like. Circle uses a ported driver and one has to be very familiar with the whole technology to write such a driver. In Circle this driver must work reliable on RPi 1-4 and Zero with different interfaces (EMMC or SDHOST) and there are a lot of different SD cards out there, which should all work. I spent a lot of time to modify this driver for Circle, for different RPi models and requirements, but I wouldn't say I'm familiar with the SD card technology. That's why I was happy, that the existing driver seems to work reliable, which was the most important requirement for me.

I understand that for your application purpose, SD card speed is important. So if you have the time to dive into SD card technology, so that you can modify the driver and can tell you have a good feeling that it will work with your modifications on any RPi model and on (say) most SD cards, I would be glad to accept these modifications for Circle.

Please note:

* The *emmc.cpp* driver is used in three different modes: With the EMMC interface of the RPi 1-3 or Zero, if USE_SDHOST is not defined; with the SDHOST interface of the RPi 1-3 or Zero, if USE_SDHOST is defined; or with the EMMC2 interface of the RPi 4. If the SDHOST interface is used, the *emmc.cpp* driver acts as high-level driver, while the low-level driver is in *sdhost.cpp*. Otherwise *emmc.cpp* is a complete driver for the EMMC or EMMC2 interfaces.

* RPi 1-3 and Zero do not support 1.8V SD card power at all and RPi 4 has (probably) a voltage switch via a GPIO expander. I was not sure about this, so I disabled 1.8V support.

dwhinham, 2020-12-17:
Thankyou for the info!

I completely understand the desire for stability/compatibility as a priority. Perhaps if we are unsure about widespread compatibility, we could have 50MHz detection/switching as an opt-in `#define` that the developer can add to Config.mk as usual.

Maybe to start with we can just ignore 1.8V on the Pi 4 and aim for 3.3V "High Speed" mode, which I think should be achievable by just about any SD card you can buy today. I have Pi 0, 1, 3B+/A+ and 4 on-hand here and SD cards of varying quality so I can test them thoroughly.

I'll get back to you with my findings soon.
Cheers!

rsta2, 2020-12-17:
Ok. Enabling this with a system option (define) is a good idea. I have also a number of RPi models and a few SD cards for testing.

dwhinham, 2020-12-28:
Let's close this and continue discussion over at #174. 🙂

closed 2020-12-28 by dwhinham