rsta2/circle #173: can't instantiate a second CLVGL class on second framebuffer (Pi4)

sky-flix, 2020-12-26:
Forgive me if this is the wrong way to report this, but I'm trying to sort out why I can't seem to instantiate a second CLVGL class on a second CScreenDevice or CBcmFrameBuffer. My first CLVGL object works fine on the primary display, and I can initialize and use the standard SetPixel methods on the second CScreenDevice, but if i create a second CLVGL (or CUGUI class for that matter), then the kernel fails to boot. I'm using the network bootloader to push the kernel8-rpi4.img as I develop, and when I upload, nothing happens. If I remove the second CLVGL or CUGUI objects, then the kernel boots and runs fine. Any thoughts as to what might be causing this?

rsta2, 2020-12-27:
This is the right place to ask. Unfortunately it's only possible to display a GUI on one frame buffer. More is not supported.

sky-flix, 2020-12-27:
Thanks for the confirmation Rene. And I appreciate everyone's work on the library. This has been a very intriguing learning experience for me.