rsta2/circle #175: Question about "do not change this order" comment

toptensoftware, 2021-01-04:
All the circle examples have the comment "do not change this order" in kernel.h before the various circle member variables.  What's the reason for this and if important how do we know the correct order for particular set of services should be?

eg:

```
private:
	// do not change this order                 <== Why? What is the correct order if I add something?
	CMemorySystem		m_Memory;
	CActLED			m_ActLED;
	CKernelOptions		m_Options;
	CDeviceNameService	m_DeviceNameService;
	CScreenDevice		m_Screen;
	CSerialDevice		m_Serial;
	CExceptionHandler	m_ExceptionHandler;
	CInterruptSystem	m_Interrupt;
	CTimer			m_Timer;
	CLogger			m_Logger;
	CUSBHCIDevice		m_USBHCI;
	CScheduler		m_Scheduler;
	CNetSubSystem		m_Net;

```

rsta2, 2021-01-04:
This comment should point out, that it can be important, which member variable is given first, second and so on, because the constructors of the members are called in the order, in which they are specified in the class declaration. Normally you should use a sample program, which matches your own requirements best and use the order of member variables, which is used in this sample. Your own application's member variables can be added to the end of this list then.

toptensoftware, 2021-01-05:
Ok, so there are startup order dependencies between some of these classes? Makes sense... thanks.

closed 2021-01-05 by toptensoftware