rsta2/circle #180: Policy for contributing breaking changes

toptensoftware, 2021-01-07:
I'm just wondering what the policy is for contributions to circle that may break existing clients?

For example: I'm need a way to switch the usb keyboard device between raw mode and not.  There's a couple of ways I can tweak the existing code to make this possible, but each would be a breaking change for existing clients (albeit a trivial one to fix).  Alternatively I can add a second api to do the same/similar thing which wouldn't affect existing clients, but it's more bloaty.

Just wondering what is your policy/approach on these kinds of changes?

rsta2, 2021-01-08:
There is no special policy for that. It depends. In this special case (switching between raw and cooked keyboard mode), which is probably not interesting for most users, I would suggest to create your own fork and manage the modifications there.

toptensoftware, 2021-01-09:
Thanks Rene...  the keyboard mode was just an example (which I managed to work around with needing changes anyway).  It just got me wondering about how I should approach any changes I might need - hopefully I won't need any :)

(btw: I've only been using Circle for about a week, but really enjoying it.  Very few issues and fits what I need almost perfectly. Thank you for all the effort you've put into building it)

closed 2021-01-09 by toptensoftware