rsta2/circle #182: Creating Tasks in a suspended state.

toptensoftware, 2021-01-09:
I've hit a problem with trying to initialize tasks. The problem is related to the fact that I'm using OpenGL and many of the methods yield the current task.  Suppose I have a constructor like this:

```
CMyTask::CMyTask()
{
    glSomeFunction();
}
```

then its `Run` method gets called before the constructor has completed because the base `CTask` constructor registers the task with the scheduler in a ready state, but the derived class constructor yields before it's finished.

I tried moving the initialization from the task constructor to the top of the `Run` method, but then I have a class that's not fully initialized and I need to wait for it to start before calling other methods on it.  I also considered moving the constructor code externally somewhere and execute it before constructing the task, but that's just super messy.  Neither options is ideal.

As an experiment, I made the `CTask::SetState` method public and tried the following which fixes the issue but is hacky.

```
CMyTask::CMyTask()
{
    SetState(TTaskState::TaskStateBlocked);
    glSomeFunction();
    SetState(TTaskState::TaskStateReady);
}
```

Is there already a way to better control the starting up tasks that I've missed?

If not, a couple of small, non-breaking changes could make this much easier:

1. Add a new Task state `TTaskStateNew`
2. Add a boolean `createSuspended` flag to the task constructor to initialize the task in the `TTaskStateNew` state instead of `TTaskStateReady`
3. Add a new task method `Start()` which asserts it's a new task and switches it to the ready state
4. New tasks obviously wouldn't be scheduled until they're started.

This could be further enhanced:

1. Add a new method to the scheduler `SuspendNewTasks()` which forces the `TTaskStateNew` for all new tasks
2. Add a second method to the scheduler `ResumeNewTasks()` which calls `Start` on all tasks in the new state.

I like the idea of suspending all new tasks as it means you don't have to worry about things accidentally starting in half initialized states while everything is getting wired up.  Taking it a step further you could include a `AbortNewTasks` to handle cases where something failed.

I'm happy to make these changes, because I think I need them (or at least really want them) but wanted to run it by you first.  Thoughts?

rsta2, 2021-01-10:
I understand the problem. Please modify the classes CTask and CScheduler and create a pull request to implement the suggested changes. Please branch off from the *develop* branch, where development takes place.

I'm not fully clear about the semantics of `SuspendNewTasks()`, because there may be tasks in the system, which are not under your control and which create other tasks by themself. These tasks expect to start in `TaskStateReady`.

toptensoftware, 2021-01-10:
Hi Rene,

I've created PR https://github.com/rsta2/circle/pull/183 that includes these changes.

> I'm not fully clear about the semantics of SuspendNewTasks(), because there may be tasks in the system, which are not under your control and which create other tasks by themself. These tasks expect to start in TaskStateReady.

You're obvious a much better judge of the impact of this, but my guess is that it shouldn't be an issue.  Usually you'd just use it across the creation of a set of related tasks and you wouldn't leave it in a long running SuspendNewTasks state.

eg: in my emulator project has three closely related tasks, so it looks like this:

```
m_Scheduler.SuspendNewTasks();
m_pTaskRender = new CTaskRender();
m_pTaskCpu = new CTaskCpu();
m_pTaskConsole = new CTaskConsole();
m_Scheduler.ResumeNewTasks();
```

Even if something in there unknowingly started other system tasks, they'd start running immediately on ResumeNewTasks() being called anyway.  One issue might be if the current task waited on an event being signalled from one of the tasks that should be starting which would block indefinitely.

rsta2, 2021-01-11:
Thanks for the PRs! I have to test this carefully, which will take some time. Your explanation on `SuspendNewTasks()` makes sense. There shouldn't be a problem with it.

rsta2, 2021-01-11:
I tested your modifications with a number of sample programs, which use the scheduler (e.g. 21-webserver, VCHIQ sound, OpenGL ES triangle sample, WLAN) and everything worked all right. I did not test the new features (deferred task start), but I suppose you have already done that. Both PR #183 and #184 have been merged to the *develop* branch. Thanks for your work!

closed 2021-01-11 by toptensoftware