rsta2/circle #185: Question about enumerating multiple device instances

toptensoftware, 2021-01-12:
Just looking at how device names are managed for multiple instances of a device (eg: mouse1, mouse2) and I'm wondering what is the the correct way to discover all attached instances.  One way would be to just query for every possible number up to some limit, but that seems messy.

Is it worth adding a method to enumerate all devices of a particular kind?  I'm happy to code it if you think it's worth adding.

toptensoftware, 2021-01-12:
I decided I really need this and have implemented the following on the `CDeviceNameService`.  I can send a PR if you like.

```cpp
	/// \brief Enumerate all devices, or all devices of a specified prefix
	/// \param pPrefix Prefix of devices to be enumerated (or null for all devices)
	/// \param callback A callback to be invoked for each matching device
	/// \param arg A user defined pointer that will be back passed to the callback
	/// \return false if the enumeration was cancelled by the callback returning false
	boolean EnumerateDevices (
		const char* pPrefix, 
		boolean (*callback)(CDevice* pDevice, const char* name, boolean bBlockDevice, void* arg), 
		void* arg
	);
```

rsta2, 2021-01-12:
While implementing USB plug-and-play, I thought about the device enumeration from application view. One option was, to add a AddDevice() callback for the application, but I decided against it. I think, most users use only one instance of a device class, which has the index 1 then. Even if one wants to enumerate more devices of a class, it's not difficult. One only have to check the indices 1 to N, where N is the number of devices, you want to support. Currently I don't want to change this, because I think, it would make Circle more complicated.

toptensoftware, 2021-01-12:
Hi Rene, no problem, I can just keep this on my fork.  I expect there might be other changes I'll need for my project going forwards. If I think they might be useful I'll send PRs but please don't feel compelled to accept them if you don't think they're needed - I actually appreciate the restraint. - Brad

closed 2021-01-12 by toptensoftware