rsta2/circle #189: Sample program for FATFS and USB storage devices doesn't seem to work

toptensoftware, 2021-01-16:
I've been trying to learn how to use USB storage devices, and was looking at sample 39.  It works fine using CFATFileSystem, but if I set USE_FATFS nothing seems to happen - inserting and removing drives does nothing... perhaps even hangs, not sure.  The drive has a single partition (not sure if that matters).  In my other test program I can see the devices being detected by USB so that bit is working.

I can dig in further but wanted to check with you to see if there's anything else that might be breaking this, and perhaps let me know if it's working for you atm.

Running from develop branch, but nothing changed.

rsta2, 2021-01-17:
The sample/39-umsdplugging from the *develop* branch does work here with `USE_FATFS`. I have tested this again on a RPi 2B with two different USB flash drives. Did you make `make clean` after setting `USE_FATFS`? Are you using the firmware downloaded in boot/?

toptensoftware, 2021-01-18:
Thanks Rene... leaving it with me and I'll take a fresh look at it again tonight.

toptensoftware, 2021-01-18:
Had another look at this and it appears to be related to the USB to SD Card adapter I'm using.  When I switched to a different (older) one it seems to work much better even with the same SD card.

Not sure if it helps, but below is some details on what seems to be going wrong.

Finally (and I'm not entirely sure about this) but it seems to work better when I put the sample 39 kernel image direcrtly on the Pi's boot SD card rather than flashing with the bootloader.  Are there cases the boot loader can't handle?  (Also, just to be clear I'm using the latest development branch with none of my recent changes for everything - a fresh git clone to a different directory). I'm using the recommended 9.2 ARM toolchain.

To answer questions: Yes, I did clean after changing `USE_FATFS` and Yes, using firmware from boot directory.  Running 32-bit arch.
---

With the poorly behaving adapter, with CFATFileSystem I get this error:

[image]

Similar with FATFS:

[image]

The adapter that's causing trouble is this:

[image]

rsta2, 2021-01-18:
I wonder, because there is no `assert (!(usChangeStatus & C_PORT_ENABLE__MASK))` in line 566 of *usbstandardhub.cpp* on the *develop* branch. It has been removed there recently. Are you testing the version from *master* be any chance?

toptensoftware, 2021-01-19:
Hi Rene,  sorry to have wasted your time.  I must have mistyped my checkout command after the fresh clone of the repo and forgot to include the remote name... so my develop branch was just a branch from master.  Silly mistake. Working fine now, thank you.

closed 2021-01-19 by toptensoftware