rsta2/circle #191: M-Audio USB Uno causes hard freeze when hotplugged on Pi 3A+

dwhinham, 2021-01-18:
Hi Rene!

I've been testing some USB MIDI devices on Step 43.2 (and also the latest `develop`) with the `miniorgan` sample, on a Raspberry Pi 3 A+.

- When the device is present on startup, everything is fine.
- When the device is removed and re-inserted, it may work, or may cause a hard freeze, or it may cause an assertion.

A possible clue: on a **Raspberry Pi 3 B+**, which of course has the integrated USB hub, I seem to be able to remove/reinsert the device several times with no issues, so perhaps this problem is unique to the Pi 3A+ having no hub?

Here's an example of the assertion case - as you can see I was able to remove it and reinsert it once, but the second time caused a crash:
[image]

Here is some information from `lsusb` on Linux:
```
Bus 001 Device 015: ID 0763:0150 M-Audio M-Audio Uno
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0763 M-Audio
  idProduct          0x0150 M-Audio Uno
  bcdDevice            1.13
  iManufacturer           1 M-Audio
  iProduct                2 USB Uno MIDI Interface
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0065
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          3 Audio Class
    bmAttributes         0xc0
      Self Powered
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0
      iInterface              0
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0009
        bInCollection           1
        baInterfaceNr(0)        1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         1 Audio
      bInterfaceSubClass      3 MIDI Streaming
      bInterfaceProtocol      0
      iInterface              0
      MIDIStreaming Interface Descriptor:
        bLength                 7
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0041
      MIDIStreaming Interface Descriptor:
        bLength                 6
        bDescriptorType        36
        bDescriptorSubtype      2 (MIDI_IN_JACK)
        bJackType               1 Embedded
        bJackID                 1
        iJack                   0
      MIDIStreaming Interface Descriptor:
        bLength                 6
        bDescriptorType        36
        bDescriptorSubtype      2 (MIDI_IN_JACK)
        bJackType               2 External
        bJackID                 2
        iJack                   0
      MIDIStreaming Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      3 (MIDI_OUT_JACK)
        bJackType               1 Embedded
        bJackID                 3
        bNrInputPins            1
        baSourceID( 0)          2
        BaSourcePin( 0)         1
        iJack                   0
      MIDIStreaming Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      3 (MIDI_OUT_JACK)
        bJackType               2 External
        bJackID                 4
        bNrInputPins            1
        baSourceID( 0)          1
        BaSourcePin( 0)         1
        iJack                   0
      Endpoint Descriptor:
        bLength                 9
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
        bRefresh                0
        bSynchAddress           0
        MIDIStreaming Endpoint Descriptor:
          bLength                 5
          bDescriptorType        37
          bDescriptorSubtype      1 (GENERAL)
          bNumEmbMIDIJack         1
          baAssocJackID( 0)       3
      Endpoint Descriptor:
        bLength                 9
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
        bRefresh                0
        bSynchAddress           0
        MIDIStreaming Endpoint Descriptor:
          bLength                 5
          bDescriptorType        37
          bDescriptorSubtype      1 (GENERAL)
          bNumEmbMIDIJack         1
          baAssocJackID( 0)       1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

Thanks!

dwhinham, 2021-01-18:
Thanks for the suggestion - with this option enabled, it seems like sometimes I am able to perform a few more unplug/plug cycles before getting a freeze/assertion, but unfortunately they still occur. The assertion is always the same as in the log screenshot.

rsta2, 2021-01-18:
Thanks for testing! Referring to the image with the log messages above, I suppose the assertion fails about 1.8 seconds after removing the device? Or is this the next plug-in cycle?

dwhinham, 2021-01-18:
Sorry, I should have clarified - it's the next plug-in cycle. The freeze or assertion will happen pretty much instantly on inserting the device.

I've done some more testing and triggering the assertion is quite rare, the hard freeze is much more common. Occasionally I might be able to get 5 or 6 successful remove/insert cycles, but usually it's a hard freeze on the first insertion, especially if I start the kernel without the device connected.

By "hard freeze", I mean that the spinner in the top-right stops spinning, and further USB plug cycles cause no effect.

It "feels" like it could be timing-related or a race condition, although I appreciate that's a very vague thing to say.

Thanks!

rsta2, 2021-01-19:
Thanks for info! I thought about it, but currently I do not have an idea, what goes wrong here. Unfortunately I do not have a M-Audio USB Uno and neither a RPi 3A+. We could try to find this out together, starting with adding some more log output to find out, where it hangs. I think, this works better using Email. I would send you a kernel image and you would report the results. Otherwise I could add this MIDI interface to the list of unsupported devices (with USB PnP) in the Wiki.

dwhinham, 2021-01-19:
> I think, this works better using Email. I would send you a kernel image and you would report the results.

No problem at all, I would be more than happy to do this if it's not too much trouble! The problem isn't a huge issue, but I thought maybe it revealed a possible underlying bug that could occur with other unusual combinations of devices.

Feel free to contact me at daleyo@gmail.com with any test kernels, and I'll respond with the results.

Thankyou!

rsta2, 2021-01-20:
For the record: We found out, that the connect state of the tested M-Audio USB Uno is not stable after attaching the device. It bounces from "connected" to "disconnected" and "connected" again in a short time. The Circle USB plug-and-play code cannot handle this.

Because it's a relatively old device, newer versions of this MIDI interface may have a different behavior. In any case the M-Audio USB Uno is possibly not hotplug-friendly, if used with Circle.

dwhinham, 2021-01-21:
Yes, I think this is an edge-case quirk of my device, which could be slightly faulty (it's a bit old now) and it's more trouble than it's worth to try and find a way to handle it in Circle.

Thanks for your help Rene, I'll mark this as closed.

closed 2021-01-21 by dwhinham