rsta2/circle #193: Additional CTask user data slot

toptensoftware, 2021-01-20:
I need the ability to associate a pointer with a CTask. (in this particular case I want to associate a message queue object with an arbitrary task object)

For my own CTask instances its easy because I can just add a member variable.  For the default task created by the scheduler I can't.

I see that CTask already has a Get/SetUserData... one extra slot there would suffice nicely.  How would you feel about either of the following:

1. A sysconfig setting to add extra slots, or
2. A permanent extra slot slot for client program use.

rsta2, 2021-01-20:
I added another user data slot for application usage to CTask. It's on the *develop* branch. If I would make this configurable and Circle would need another slot in the future by itself, it would be not clear, which slot some application would already use and where to add the new slot.

toptensoftware, 2021-01-21:
Perfect!  Thanks Rene.

closed 2021-01-21 by toptensoftware