rsta2/circle #195: Busy spinning during disk transfers

toptensoftware, 2021-01-21:
I've been looking at how disk access in Circle works and was a little surprised to find that the SD card and USB storage devices both enter tight CPU spin loops while waiting for transfers to finish. I can understand why as I'm sure going re-entrant into these functions would be a disaster.  On the other hand, waiting for a device is a perfect opportunity to hand off to another task for a bit. So I'm wondering...

Firstly, is my assumption about re-entrancy with these devices correct?  ie: 

* it _is not_ safe to read/write from multiple cores or from irq handlers 
* it _is_ safe to read/write from multiple tasks (since they can't go re-entrant).

Second, it would be nice if these devices could yield to the scheduler while waiting.  Obviously the danger here is that it would make re-entrancy from single core possible and would be a fairly risky change to impose on all Circle clients. 

However, what if this was plugged in with a "synchronization context" that's passed to the device constructor:

```cpp
struct ISynchronizationContext
{
    virtual void Acquire() = 0;
    virtual void Release() = 0;
    virtual void Yield() = 0;
};
```

and the devices would call `Acquire`/`Release`  at the top-level when entering a read/write operation and `Yield` from inside their current busy spin loops.  You could then implement a default blocking/non-reentrant context that wouldn't affect current semantics:

```cpp
class CBusySpinContext : public ISynchronizationContext
{
    virtual void Acquire() override { return; }
    virtual void Release() override { return; }
    virtual void Yield() override { return; }
};
```

But adventurous clients could opt-in to a yielding version:

```cpp
class CYieldingContext : public ISynchronizationContext
{
    virtual void Acquire() override { m_mutex.Acquire(); }
    virtual void Release() override { m_mutex.Release(); }
    virtual void Yield() override { CScheduler::Get()->Yield(); }
    CSynchronizationMutex m_mutex;    // See PR: 187
}
```

Thoughts? 

PS: I'm about to reach a point in my project where I'll have multiple tasks performing disk access... I can do some experiments if you like.

rsta2, 2021-01-21:
Yes, that's not perfect for the system throughput, but the scheduler is optional in Circle and USB and the SD card driver must be able to run without it. Your synchronization context suggestion, handed over to the constructor, may be a solution for this. I will deal with this, when the next Circle release is done. Please don't send a pull request.

toptensoftware, 2021-01-21:
Hi Rene,  yes keeping the scheduler decoupled was part of what I was thinking with this too. Look forward to whatever you come up with.

rsta2, 2021-02-01:
You closed this issue. Does this mean, you are not interested in this any more?

toptensoftware, 2021-02-02:
Hi Rene, yes, still interested in this. I closed it because it sounded like something that was on your radar to do anyway.  Re-opened it to flag interest.

rsta2, 2021-02-02:
Ah, OK. I usually let an issue open, until it is solved or it is clear, that it won't be solved.

rsta2, 2021-02-06:
There is a solution for the busy wait problem on the *develop* branch. You have to define the system option `NO_BUSY_WAIT` to enable it. Be sure that the scheduler object is constructed before the CEMMCDevice and CUSBHCIDevice objects.

On the Raspberry Pi 1-3 and Zero the SDHOST driver is used by default for SD card access. This driver does the I/O transfers from the IRQ handler, which is not perfect, if used together with `NO_BUSY_WAIT`. That's why you should define the system option `NO_SDHOST` too for better system throughput. Unfortunately this option cannot be used, if you want to have WLAN access.

toptensoftware, 2021-02-08:
Hey Rene,  thank you! 

(Not sure I understand the implications of those configuration options, but I'll check it out).

rsta2, 2021-02-08:
`NO_BUSY_WAIT` suppresses the busy waiting in the SD card and USB drivers as requested by this issue. I implemented this different than expected, because I found the "synchronization context" solution too complicated, while implementing it.

Also the assumption, that transfers cannot be initiated from secondary CPU cores (1-3), wasn't right. Previously this was allowed, but with `NO_BUSY_WAIT` defined the application developer asserts, that transfers will be only initiated from core 0, so `CScheduler::Get()->Yield()` can be called instead of just busy spinning.

`NO_SDHOST` uses the EMMC driver instead of the SDHOST driver for the SD card, which can be accessed using two different interfaces on the RPi 1-3 and Zero. EMMC was the first solution, but is used for the WLAN too on these RPi models. That's why I ported the SDHOST driver from Linux and enabled it by default. Unfortunately this driver does not work perfectly with `NO_BUSY_WAIT`, why I suggested to define the `NO_SDHOST` option.

rsta2, 2021-05-14:
This feature is implemented in Circle 44. So this can be closed.

closed 2021-05-14 by rsta2