rsta2/circle #197: Truncated assert messages when logging via serial

toptensoftware, 2021-01-21:
I've noticed that when using serial port for monitoring, assert notifications are truncated, or don't appear at all and tracked it down to this:

https://github.com/rsta2/circle/blob/d069899db50011aa286f7ece04e3249dba0316bf/lib/logger.cpp#L166

Replacing the `halt()` call with `while (1);` fixes the issue. 

Also, halt causes magic reboot handler to not work... whereas with the while loop it does.

Or perhaps there's another solution for this?

rsta2, 2021-01-21:
Yes, `halt()` disables all interrupts, so if you use `CSerialDevice` with IRQ or FIQ and direct the logger output to it, you should install a panic handler with `CLogger::RegisterPanicHandler()`, which never returns. This has the same effect as replacing `halt()` with `while(1)`.

toptensoftware, 2021-01-21:
OK thanks.

closed 2021-01-21 by toptensoftware