rsta2/circle #199: QEMU Build Fails on Ubuntu 20.04

toptensoftware, 2021-01-25:
I was trying to follow the [instructions here](https://github.com/rsta2/circle/blob/master/doc/qemu.txt) for setting up QEMU environment but the build fails with the errors shown below (running on Ubuntu 20.04)

Is this still the recommended way to run Circle projects under QEMU and if so, what version of Linux/Ubuntu do I need to build it?

~~~
/home/brad/Projects/qemu/qga/main.c:274:5: error: ‘GTimeVal’ is deprecated: Use 'GDateTime' instead [-Werror=deprecated-declarations]
  274 |     GTimeVal time;
      |     ^~~~~~~~
In file included from /usr/include/glib-2.0/glib/galloca.h:32,
                 from /usr/include/glib-2.0/glib.h:30,
                 from /home/brad/Projects/qemu/include/glib-compat.h:19,
                 from /home/brad/Projects/qemu/include/qemu/osdep.h:100,
                 from /home/brad/Projects/qemu/qga/main.c:13:
/usr/include/glib-2.0/glib/gtypes.h:547:8: note: declared here
  547 | struct _GTimeVal
      |        ^~~~~~~~~
/home/brad/Projects/qemu/qga/main.c:289:9: error: ‘g_get_current_time’ is deprecated: Use 'g_get_real_time' instead [-Werror=deprecated-declarations]
  289 |         g_get_current_time(&time);
      |         ^~~~~~~~~~~~~~~~~~
In file included from /usr/include/glib-2.0/glib/giochannel.h:33,
                 from /usr/include/glib-2.0/glib.h:54,
                 from /home/brad/Projects/qemu/include/glib-compat.h:19,
                 from /home/brad/Projects/qemu/include/qemu/osdep.h:100,
                 from /home/brad/Projects/qemu/qga/main.c:13:
/usr/include/glib-2.0/glib/gmain.h:679:8: note: declared here
  679 | void   g_get_current_time                 (GTimeVal       *result);
      |        ^~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
~~~

rsta2, 2021-01-25:
This QEMU fork is relatively old and building it produces many warnings, which will be treated as errors by default, which aborts the build. Please add the option `--disable-werror`, when calling `configure`. It should be possible to complete the build then.

toptensoftware, 2021-01-27:
OK... I'll give that a go.

closed 2021-01-27 by toptensoftware