rsta2/circle #200: Serial port in interrupt mode dropping data.

toptensoftware, 2021-01-27:
When using the serial port in interrupt mode for logging, data will be lost if written too quickly - presumably due to the output buffer filling up.

Is this expected and what's the correct way to deal with it? 

fwiw: I'm trying to print out the results of a bunch of unit tests (saves having to attach a monitor), but part way through it starts dropping lots of data.  Doesn't happen if not using interrupt driven serial port.

toptensoftware, 2021-01-27:
I managed to hack this to work, but I'm hoping there's a better way...

Changed this:

Here: https://github.com/rsta2/circle/blob/1d7d3e6526f1b3fa60a73f3419a59a49f23a8f96/lib/serial.cpp#L350

To this:

```cpp
	while (nCount--)
	{
		while (!Write (*pChar))
		{
			m_LineSpinLock.Release ();
			m_LineSpinLock.Acquire ();
		}

		if (*pChar++ == '\n')
		{
			if (m_nOptions & SERIAL_OPTION_ONLCR)
			{
				while (!Write ('\r'))
				{
					m_LineSpinLock.Release ();
					m_LineSpinLock.Acquire ();
				}
			}
		}

		nResult++;
	}
```

rsta2, 2021-01-27:
Normally I use the non-interrupt driven variant for logging much data. You can also increase the serial buffer size [here](https://github.com/rsta2/circle/blob/master/include/circle/serial.h#L54).

toptensoftware, 2021-01-29:
Thanks Rene.  I ended up reverting my hack and made a better fix by overriding `CSerialDevice::Write()` in a my own class and having it check the returned count from CSerialDevice::Write, and everything wasn't written it yields and then tries again.

closed 2021-01-29 by toptensoftware