rsta2/circle #201: Wrong button count reported for Xbox 360 USB controller

rhester72, 2021-01-27:
This line:

#define REPORT_BUTTONS			16

in lib/usb/usbgamepadxbox360.cpp should actually have the value 17, as that is the actual number of digital buttons available.  Otherwise, the last button definition (d-pad left) is inaccessible.

rsta2, 2021-01-27:
The [Buttons field](https://github.com/rsta2/circle/blob/master/lib/usb/usbgamepadxbox360.cpp#L39) in the gamepad report has only 16 bits, so changing this value to 17 has no positive effect. There may be different variants of this gamepad. On the one I have here, the D-pad left button is working.

rhester72, 2021-01-27:
A 16-bit map _could_ be sufficient for the gamepad report, as there are (easily demonstrable) 17 buttons (0 and 3-18, 1 and 2 are unmapped), and 0 could be considered a valid value for said map (corresponding to the Xbox button, which is 0), though that'd be a bit hokey.  BMC64 (a Circle-based emulator) uses the passed REPORT_BUTTONS count (in the num_buttons array) as a high water mark for how many buttons to present in the assignment UI.  As Circle reports only 16, the last one (dpad left) cannot be assigned (within BMC64) without this change.

Please note this does *NOT* affect reporting of _button presses_...this is ONLY an issue for applications based on Circle that depend (for UI purposes, for example) on an accurate button count from Circle independent of registering actual button presses.

rsta2, 2021-01-28:
The number of digital buttons is returned in the [nbuttons field](https://github.com/rsta2/circle/blob/master/include/circle/usb/usbgamepad.h#L153) of the `TGamePadState` struct, which can be fetched using `CUSBGamePadDevice::GetInitialState()`. This field is set [here](https://github.com/rsta2/circle/blob/master/lib/usb/usbgamepadxbox360.cpp#L79) for this gamepad. The resulting value is 18, because there are two analog buttons, which can be read as digital buttons too.

rhester72, 2021-01-28:
Right, nbuttons is what's being populated with 18 (16 digital + 2 analog 'buttons'), which is wrong - by that token, it's only 15 + 2 (Xbox/Back/Start/A/B/X/Y/L1/L2/R1/R2/U/R/D/L digital and L3/R3 analog).  Unfortunately, this one is REALLY strange because from a button number standpoint, 1 and 2 are unused.  The layout looks like this:

0  Xbox
1  UNUSED
2  UNUSED
3  L2
4  R2
5  L1
6  R1
7  Y
8  B
9  A
10 X
11 Back
12 L3
13 R3
14 Start
15 U
16 R
17 D
18 L

and therein lies the problem - for apps that require the nbuttons field to be accurate, they expect a value of 17 (15+2), _and_ they expect the buttons to be sequentially numbered from 0..16, which they are not.  Since I have to imagine that creating code to detect the 'unused' buttons and remap those above it to slide "down" into those empty slots would be a LOT more pain (and confusing) than it's worth, the only other realistic solution is to "lie" about the button count, and present it as 19 (17+2), which accounts for the two unused slots and allows the app using nbuttons to correctly consider button numbers 0..18.

rsta2, 2021-01-29:
Now I understand the problem. I will fix this soon. Thanks for explaining this!

rsta2, 2021-01-29:
The fix is on the *develop* branch for the moment. Thanks for reporting this issue.

rsta2, 2021-05-14:
This issue has been fixed in Circle 44. So this can be closed.

closed 2021-05-14 by rsta2