rsta2/circle #204: Circle based programs on BerryBoot?

msx80, 2021-02-28:
Hi there, a user recently asked if my circle based program could run on [BerryBoot](https://www.berryterminal.com/doku.php/berryboot), which is a multi bootloader for RPi that lets you install different OSes and choose one at startup. I gave a quick look yesterday but it's nothing obvious.

Before eventually diving in, is there any prior knowledge about running circle based "OSes" along others with BerryBoot? Perhaps someone already did it or can rule out that it's possible.

They have an [Adding your own custom operating systems](https://www.berryterminal.com/doku.php/berryboot/adding_custom_distributions) page but i couldn't get anything out for this case.

Thanks :)

rsta2, 2021-03-01:
As I read it, BerryBoot only works with Linux-based OS distributions.

rsta2, 2021-04-04:
There was no clue for over a month now. I suppose, BerryBoot does not work with Circle apps, and will close this.

closed 2021-04-04 by rsta2

msx80, 2021-04-06:
ah sorry for not reporting, i also think it works only on linux based OSes, i haven't looked into it anymore.

rsta2, 2021-04-07:
OK, no problem!