rsta2/circle #205: 8BitDo USB gamepad issue

flabbergast, 2021-03-01:
I have an [8BitDo SN30 pro USB](https://www.8bitdo.com/sn30-pro-usb-gamepad/) gamepad. It does not work correctly with Circle and I am stumped. I would appreciate some pointers, but am also happy to close this if it turns out too non-standard/fiddly.

Running sample/27-usbgamepad on a (headless) Raspberry Pi Zero sees the gamepad enumerate correctly (it pretends to be an Xbox 360 gamepad). This is the develop branch, but vanilla 43.3, 42.1 are essentially the same.
```
logger: Circle 43.3 started on Raspberry Pi Zero
00:00:01.00 timer: SpeedFactor is 1.00
00:00:01.86 usbdev0-1: Device ven45e-28e found
00:00:01.87 usbdev0-1: Using device/interface ven45e-28e
00:00:01.87 usbdev0-1: Function is not supported
00:00:01.88 usbdev0-1: Function is not supported
00:00:01.88 usbdev0-1: Function is not supported
00:00:01.94 dwroot: Device configured
00:00:01.94 kernel: Compile time: Mar  1 2021 19:08:20
00:00:01.95 kernel: Please attach an USB gamepad, if not already done!
00:00:01.95 kernel: Gamepad 1: 19 Button(s) 0 Hat(s)
00:00:01.96 kernel: Gamepad 1: Axis 1: Minimum 0 Maximum 255
00:00:01.96 kernel: Gamepad 1: Axis 2: Minimum 0 Maximum 255
00:00:01.97 kernel: Gamepad 1: Axis 3: Minimum 0 Maximum 255
00:00:01.97 kernel: Gamepad 1: Axis 4: Minimum 0 Maximum 255
00:00:01.98 kernel: Gamepad 1: Axis 5: Minimum 0 Maximum 255
00:00:01.98 kernel: Gamepad 1: Axis 6: Minimum 0 Maximum 255
00:00:01.99 kernel: Use your gamepad controls!
```
However, no button press events are received at all. Disconnect event works fine.

I played a bit with the code, and...
- the gamepad responds fine to SetLEDMode and SetRumbleMode
- the execution never seems to get into the ReportHandler
- the curious part: I recently "ported" your Xbox 360 gamepad driver back to uspi, and there it "works" ... but the USB events are appearing continuously, even when nothing changes. But the button presses/axes are reported correctly.
- I have another "pretend Xbox 360 gamepad" (inside a [GPi Case](http://retroflag.com/GPi-CASE.html)), and that one behaves OK: enumerates fine (same vid/pid), events appear (only) when buttons are pressed.

All in all, it seems that there is something wrong with the 8BitDo controller itself, regarding the USB communication with dwhci. I have two, both act the same. They work fine in linux. Was just wondering if you'd have an idea for a workaround in Circle.

rsta2, 2021-03-02:
Unfortunately it's always difficult to react on problem reports for gamepads, which should be compatible with a Circle-supported gamepad model, but do not work with it. It's not possible for me to order all these devices, because there are that many different out there. But without having the hardware here, I cannot do much.

Because you ported the gamepad Circle code to USPi, you already have some experience with it and perhaps you can investigate this on your own. One thing you can try is adding the following line to *Config.mk* and rebuild all:

```
DEFINE += -DNO_USB_SOF_INTR
```

This changes the behavior of the USB host controller driver, so that it is compatible with that of USPi. You wrote, that your ported driver is working with USPi.

Another thing to try is enabling the debug output at [this line](https://github.com/rsta2/circle/blob/master/lib/usb/usbdevice.cpp#L319). This dumps the USB configuration descriptor of the attached device(s). If you would post this output here, I can have a look on it, if it is all right.

If this doesn't help, I can only add the gamepad to the list of unsupported devices in the Wiki.

flabbergast, 2021-03-02:
Thank you for the suggestions! Of course I understand that it's impossible to deal with all the quirky/bad USB devices... and there is a lot of them around. I really appreciate your effort!

I will continue to investigate, but I don't have much (any, really) host-side USB experience. I was sort of hoping that you would have seen something like this before...

In any case, here is some info. Compiling with SOF disabled yielded no change in behaviour. The descriptor:
```
00:00:01.88 usbdev: 88A0: 09 02 8B 00 04 01 00 A0-FA 09 04 00 00 02 FF 5D
00:00:01.88 usbdev: 88B0: 01 00 11 21 10 01 01 25-81 14 03 03 03 04 13 02
00:00:01.89 usbdev: 88C0: 08 03 03 07 05 81 03 20-00 04 07 05 02 03 20 00
00:00:01.90 usbdev: 88D0: 08 09 04 01 00 02 FF 5D-03 00 1B 21 00 01 01 01
00:00:01.90 usbdev: 88E0: 83 40 01 04 20 16 85 00-00 00 00 00 00 16 05 00
00:00:01.91 usbdev: 88F0: 00 00 00 00 00 07 05 83-03 20 00 02 07 05 04 03
00:00:01.92 usbdev: 8900: 20 00 04 09 04 02 00 01-FF 5D 02 00 09 21 00 01
00:00:01.92 usbdev: 8910: 01 22 86 07 00 07 05 86-03 20 00 10 09 04 03 00
00:00:01.93 usbdev: 8920: 00 FF FD 13 04 06 41 00-01 01 03 B8 55 55 55 55
```
Here is a log from uspi 2.0 (the code is in [this repo](https://bitbucket.org/fo0o/abbaye/src/gpi/), that's the 'port', but ... the non-uspi parts are not very nice, it's mostly the old "Abbaye" code of M Maccaferri, and my mess on top).
```
uspi: Initializing USPi library 2.00
usbdev0-1: Device ven45e-28e found
usbdev: Using device/interface ven45e-28e
extrausbinfo: next 'not supported' is about ven45e-28e nfunc1 dtype4 ifnum1 endp#2 class255 subcl93 prot3
usbdev0-1: Function is not supported
extrausbinfo: next 'not supported' is about ven45e-28e nfunc1 dtype4 ifnum2 endp#1 class255 subcl93 prot2
usbdev0-1: Function is not supported
extrausbinfo: next 'not supported' is about ven45e-28e nfunc1 dtype4 ifnum3 endp#0 class255 subcl253 prot19
usbdev0-1: Function is not supported
dwroot: Device configured
uspi: USPi library successfully initialized
main: GamePad 1: Vendor 0x45E Product 0x28E Version 0x114
main: GamePad 1: Manufacturer "Controller" Product "Controller"
dwhci: Transaction failed (status 0xA)
main: GamePad 1: 18 Buttons 0 Hats
main: GamePad 1: Axis 1: Minimum 0 Maximum 255
...
```
The interesting bit is "Transaction failed", which comes from [this place in uspi](https://bitbucket.org/fo0o/abbaye/src/55ecc321d93fc27466992049b5199b69bf6c65ff/kernel/uspi/dwhcidevice.c#lines-998).
I consistently get it on every run, at the same time; both on reset and on power up. Circle doesn't seem to get to the analogous place... also not sure what to dump to learn more about what sort of transaction is the strange one. In any case, uspi then seems to get events (a lot of them).

rsta2, 2021-03-02:
Thanks for testing and for the infos! The USB configuration descriptor should be OK. These "Transaction failed" messages sometimes occur, but should not disturb further function. But there is one important difference between Circle and USPi, Circle and the sample/27-usbgamepad support USB plug-and-play now. USPi does not support this.

Can you please try the following:

* Remove [this line](https://github.com/rsta2/circle/blob/master/lib/usb/usbhiddevice.cpp#L229). In case a "Transaction failed" condition occurs with USB PnP, this line ensures, that the driver shutdowns the communication with the gamepad, because this happens, when the device is removed. If a a "Transaction failed" condition would occur without removing the gamepad, this can cause the gamepad to stop functioning.

* Remove lines 104-106 [here](https://github.com/rsta2/circle/blob/master/lib/usb/usbgamepadxbox360.cpp#L104). This is a check, if the report from the gamepad is valid, but maybe the report from your gamepad is somewhat different in length or header. The driver would ignore any report then. You can also enable the the `debug_hexdump()` in line 109 to see, if any report is coming from your gamepad.

flabbergast, 2021-03-03:
Thank you. After some goose chases and red herrings (btw I am really grateful that log messages get properly queued also from ISRs), I finally figured out what made the difference: [this](https://github.com/rsta2/circle/blob/master/lib/usb/usbdevice.cpp#L338). Apparently the 8BitDo gamepad does not initialise properly if it doesn't get asked for the strings. (I suppose there is some error in the state machine in the controller firmware.) After changing that line to "#if 1", the gamepad works fine with Circle (with the default configuration).

That resolves my problem; thank you for your help! I don't think this warrants any change in your library code; perhaps a remark in the docs somewhere if you think it would be useful.

rsta2, 2021-03-03:
This is an unusual problem, but it's good, that you have found the reason. Thanks for info!

rsta2, 2021-03-12:
The manufacturer and product USB strings will be fetched from an USB device by default now on the *develop* branch. Hopefully the 8BitDo gamepad will work with the next release without modification then.

flabbergast, 2021-03-12:
Thank you very much, this is very kind of you!

closed 2021-03-12 by flabbergast