rsta2/circle #206: Cannot use placement new

leiradel, 2021-03-03:
First of all, thanks for this awesome project, I could go from 0 to a running baremetal application is just a few minutes.

I wanted to use placement new but if I add `#include <new>` to a compilation unit, it'll fail with the following error:

```
$ make
make: *** No rule to make target 'new', needed by 'frontend/Frontend.o'.  Stop.
```

From what I could Google, placement new is provided by the compiler and there's no way to implement it in code because it calls a constructor and that's not possible to do explicitly in C++.

Is it possible to use placement new with Circle?

Thanks.

rsta2, 2021-03-03:
Thanks! Placement new was unfortunately not supported by Circle so far, but I added this now on the *develop* branch. You have to `#include <circle/new.h>` to be able to use it. Can you please test, if this works for you?

leiradel, 2021-03-04:
Wow that was fast. Works like a charm. Thank you. I wish I could just `#include <new>` so it would work the same if I'm building it for Linux, but that's minor.

rsta2, 2021-03-04:
You are welcome. For compatibility with the C++ standard library in Circle there is the [circle-stdlib project](https://github.com/smuehlst/circle-stdlib), where you should be able to use `#include <new>`. I cannot provide this in Circle itself, because then there would be two include files with the same name and it's not clear, which one will be used. Thanks for testing!

leiradel, 2021-03-04:
Ah nice, I'll look into circle-stdlib later. Thanks again.

rsta2, 2021-05-14:
This feature has been implemented in Circle 44. So this can be closed.

closed 2021-05-14 by rsta2