rsta2/circle #207: Cannot construct an object as a static variable in a function

leiradel, 2021-03-04:
I have a function to create a singleton of a class, but the linker complains about some missing functions.

```cpp
lrcpp::Frontend& lrcpp::Frontend::getInstance() {
    static uint8_t fsmMemory[sizeof(CoreFsm)] alignas(CoreFsm);
    static Frontend singleton(fsmMemory);

    return singleton;
}
```

```
$ make clean && make
  CLEAN  /home/leiradel/Develop/circle/sample/libretro
  CPP   frontend/Main.o
  CPP   frontend/Kernel.o
  CPP   frontend/Frontend.o
  CPP   frontend/Fifo.o
  CC    core/main.o
  CPP   lrcpp/src/Components.o
  CPP   lrcpp/src/Core.o
  CPP   lrcpp/src/CoreFsm.o
  CPP   lrcpp/src/Frontend.o
  LD    kernel8.elf
aarch64-linux-gnu-ld: lrcpp/src/Frontend.o: in function `lrcpp::Frontend::getInstance()':
/home/leiradel/Develop/circle/sample/libretro/lrcpp/src/Frontend.cpp:65: undefined reference to `__cxa_guard_acquire'
aarch64-linux-gnu-ld: /home/leiradel/Develop/circle/sample/libretro/lrcpp/src/Frontend.cpp:65: undefined reference to `__cxa_guard_release'
make: *** [../../Rules.mk:155: kernel8.img] Error 1
```

These functions prevent the static variables from being initialized more than once should two threads try to get the singleton at the same time. I'm using an `#idef` to special case it when building with Circle, but I think it would be better if Circle supported it directly.

rsta2, 2021-03-04:
This is supported too, if you use the [circle-stdlib project](https://github.com/smuehlst/circle-stdlib). In Circle itself you can add the following to your project *Makefile* before the include statement for *Rules.mk* to suppress the errors:

```
CPPFLAGS += -fno-threadsafe-statics
```

Because Circle doesn't have a preemptive scheduler, this should work.

leiradel, 2021-03-04:
> Because Circle doesn't have a preemptive scheduler, this should work.

But it's a problem if you use multiple cores.

> This is supported too, if you use the circle-stdlib project.

It seems I really need to integrate circle-stdlib into my project :)

rsta2, 2021-03-05:
> > Because Circle doesn't have a preemptive scheduler, this should work.
>
> But it's a problem if you use multiple cores.

That's true. Unfortunately the precise semantics of `__cxa_guard_*()` is currently not that clear to me, so that I can implement it in Circle. If is is possible to implement it, it will take longer in any way.

leiradel, 2021-03-05:
I understand. There's a [open source implementation](https://opensource.apple.com/source/libcppabi/libcppabi-14/src/) (cxa_guard.cxx) that maybe could be used, but that's not critical and entire up to you. The license is [this one](https://releases.llvm.org/2.8/LICENSE.TXT) I believe.

For now I just have the `#ifdef` I mentioned above and it's working just fine. Feel free to close this issue if you're not planing on implementing this in Circle itself, and thanks for pointing me to circle-stdlib.

rsta2, 2021-03-06:
Thanks for the link! I will see, what I can do.

rsta2, 2021-03-21:
This feature has been implemented on the *develop* branch now.

rsta2, 2021-05-14:
This feature has been implemented in Circle 44. So this can be closed.

closed 2021-05-14 by rsta2