rsta2/circle #209: USB appears to be broken on Raspberry Pi 4B since Circle Step 43.3

dwhinham, 2021-03-07:
Since Circle Step 43.3 the XHCI device fails to initialize on my Raspberry Pi 4 B (4GB model).
 
The log shows the following:
[image]

My test setup is an AArch64 kernel (`39-umsdplugging` sample) compiled with the latest recommended toolchain (GCC 10) and the latest boot files including all of the .dtbs.

Reverting to Step 43.2, the GCC 9-based toolchain, and boot files associated with 43.2, everything works as expected again.
Perhaps there was a slight regression when adding support for the CM4?

Thanks for your help!

rsta2, 2021-03-08:
I tested USB before the release on a RPi 400 4GB and RPi 4B 2GB successfully, but I will test again. Do you use the firmware files, downloaded from the boot/ directory by default or have use modified the Makefile there to get the latest firmware version?

dwhinham, 2021-03-08:
Hi Rene,

Yes - I always use the version retrieved by the boot/ Makefile, so in the case of 43.3, it'd be `e15ef4e4fe8be99cd816cec901d977224a1eb07e`.

rsta2, 2021-03-08:
That's really odd. I tested the sample/39-umsdplugging again from Circle 43.3 AArch64 on a RPi 4B 2GB and RPi 400 4GB with the firmware version, you have noted. I also tested the cut-down version (gpu_mem=16). I used GCC 10 for build. Everything was working all right.

Is this RPi 4B, that you are using, a model with on-board EEPROM for the VL805 USB firmware? If so, which VL805 firmware are you using? I'm still using the vl805-000138a1.bin version on the RPi 4B 2GB.

dwhinham, 2021-03-08:
My Pi 4B is a Raspberry Pi 4 Model B Rev 1.1 according to /proc/cpuinfo, and this was the output of `rpi-eeprom-update` on RPi OS - it has the dedicated EERPROM:

```
pi@raspberrypi:~ $ sudo rpi-eeprom-update -a
BCM2711 detected
Dedicated VL805 EEPROM detected
*** INSTALLING EEPROM UPDATES ***
Checking for updates in /lib/firmware/raspberrypi/bootloader/default
Use raspi-config to select either the default-production release or latest update.
BOOTLOADER: update available
CURRENT: Mon  6 Jul 11:48:15 UTC 2020 (1594036095)
LATEST: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
RELEASE: default
VL805: update available
CURRENT: 000137ad
LATEST: 000138a1
BOOTFS /boot
EEPROM updates pending. Please reboot to apply the update.
 ```
 
I updated the VL805 firmware and bootloader, so now we have:
```
pi@raspberrypi:~ $ sudo rpi-eeprom-update
BCM2711 detected
Dedicated VL805 EEPROM detected
Checking for updates in /lib/firmware/raspberrypi/bootloader/default
Use raspi-config to select either the default-production release or latest update.
BOOTLOADER: up-to-date
CURRENT: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
LATEST: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
RELEASE: default
VL805: up-to-date
CURRENT: 000138a1
LATEST: 000138a1
```

I remembered I'd once been testing network boot, which meant my bootloader config was different (BOOT_ORDER set to 0xF21). I've also reset this to defaults by getting the default config from the firmware file and ensuring my bootloader is flashed with the same (BOOT_ORDER is now 0xF41).

So far, this hasn't changed anything.

Here's my complete build steps as a "santiy check":

Checked-out at `Step43.3`:
```
dale@StroggII:~/git/circle$ git status
HEAD detached at Step43.3
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   Rules.mk

no changes added to commit (use "git add" and/or "git commit -a")
```

Adjust `AARCH` and `RASPPI`:
```
dale@StroggII:~/git/circle$ git diff
diff --git a/Rules.mk b/Rules.mk
index 4402dbe..a56b169 100644
--- a/Rules.mk
+++ b/Rules.mk
@@ -23,8 +23,8 @@ CIRCLEHOME ?= ..
 -include $(CIRCLEHOME)/Config.mk
 -include $(CIRCLEHOME)/Config2.mk      # is not overwritten by "configure"
 
-AARCH   ?= 32
-RASPPI  ?= 1
+AARCH   ?= 64
+RASPPI  ?= 4
 PREFIX  ?= arm-none-eabi-
 PREFIX64 ?= aarch64-none-elf-
 ```

GCC version:
```
dale@StroggII:~/git/circle$ aarch64-none-elf-gcc -v
Using built-in specs.
COLLECT_GCC=aarch64-none-elf-gcc
COLLECT_LTO_WRAPPER=/home/dale/gcc-arm-10.2-2020.11-x86_64-aarch64-none-elf/bin/../libexec/gcc/aarch64-none-elf/10.2.1/lto-wrapper
Target: aarch64-none-elf
Configured with: /tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/src/gcc/configure --target=aarch64-none-elf --prefix=/tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/build-aarch64-none-elf/install// --with-gmp=/tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/build-aarch64-none-elf/host-tools --with-mpfr=/tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/build-aarch64-none-elf/host-tools --with-mpc=/tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/build-aarch64-none-elf/host-tools --with-isl=/tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/build-aarch64-none-elf/host-tools --disable-shared --disable-nls --disable-threads --disable-tls --enable-checking=release --enable-languages=c,c++,fortran --with-newlib --with-pkgversion='GNU Toolchain for the A-profile Architecture 10.2-2020.11 (arm-10.16)' --with-bugurl=https://bugs.linaro.org/
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.2.1 20201103 (GNU Toolchain for the A-profile Architecture 10.2-2020.11 (arm-10.16))
```

Build:
`dale@StroggII:~/git/circle$ ./makeall clean && ./makeall -j && make -C sample/39-umsdplugging -j && make -C boot && make -C boot/armstub armstub8-rpi4.bin`

All `*.dtb`, `*.dat*, `*.elf`, `*.bin` copied to blank SD card, as well as `armstub8-rpi4.bin` and `config64.txt` renamed to `config.txt`, and `kernel8-rpi4.img`.

This is the resulting build - does it work on your Pi 4?
[pi4-test.zip](https://github.com/rsta2/circle/files/6101566/pi4-test.zip)

For me, I get the log as shown in my original post.

rsta2, 2021-03-08:
Yes, your test build works on both RPi 4B 2GB and RPi 400 4GB, that I have here. I simply copied the files from the archive to an empty SD card and started the RPis with it.

I'm using the same bootloader, VL805 firmware and toolchain too. I cannot see any problem in your logs. The custom armstub8-rpi4.bin is not in your archive, but this should not be a problem, because the FIQ is not used, and it is working here without it.

Do you have a serial USB interface, so that you can save the logging output? Then you could remove the comment from the following lines and build and try again:

https://github.com/rsta2/circle/blob/master/include/circle/usb/xhciconfig.h#L25
https://github.com/rsta2/circle/blob/master/include/circle/usb/xhciconfig.h#L26

Perhaps I can see something in the xHCI logging output.

dwhinham, 2021-03-08:
Oops, yes - I forgot to include the armstub, but it was on the SD card - sorry about that!

Yes - I have serial console and also a JTAG debugging setup available.
Here's the serial console log with the `#define`s enabled:
```
logger: Circle 43.3 started on Raspberry Pi 4 Model B (AArch64)
00:00:00.66 timer: SpeedFactor is 1.51
00:00:00.73 pcie: Link up, 5.0 Gbps x1
00:00:00.73 xhci: 32 slots, 5 ports, 31 scratchpad bufs
00:00:01.05 xhcicmd: Execute command 9 (control 0x2400)
00:00:04.05 xhcicmd: Command timed out (control 0x2400)
00:00:04.05 xhcidev: Cannot enable slot (-2)
00:00:04.05 xhciroot: Port 1: Status 0x40200E03, Info 0x0
00:00:04.06 xhciroot: Port 2: Status 0x2A0, Info 0x0
00:00:04.06 xhciroot: Port 3: Status 0x2A0, Info 0x0
00:00:04.07 xhciroot: Port 4: Status 0x2A0, Info 0x0
00:00:04.07 xhciroot: Port 5: Status 0x2A0, Info 0x0
00:00:04.08 xhcievt: Count 64, Dequeue 0, Cycle 1
00:00:04.08 xhcievt: Dumping 0x400 bytes starting at 0x600880
00:00:04.09 xhcievt: 0880: 00 00 00 01 00 00 00 00-00 00 00 01 01 88 00 00
00:00:04.09 xhcievt: 0890: 00 00 00 01 00 00 00 00-00 00 00 01 01 88 00 00
00:00:04.10 xhcievt: 08A0: 00 19 60 00 00 00 00 00-00 00 00 01 01 84 00 01
00:00:04.11 xhcievt: 08B0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.11 xhcievt: 08C0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.12 xhcievt: 08D0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.13 xhcievt: 08E0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.13 xhcievt: 08F0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.14 xhcievt: 0900: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.15 xhcievt: 0910: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.15 xhcievt: 0920: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.16 xhcievt: 0930: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.17 xhcievt: 0940: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.17 xhcievt: 0950: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.18 xhcievt: 0960: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.19 xhcievt: 0970: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.19 xhcievt: 0980: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.20 xhcievt: 0990: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.21 xhcievt: 09A0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.21 xhcievt: 09B0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.22 xhcievt: 09C0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.23 xhcievt: 09D0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.23 xhcievt: 09E0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.24 xhcievt: 09F0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.25 xhcievt: 0A00: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.25 xhcievt: 0A10: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.26 xhcievt: 0A20: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.27 xhcievt: 0A30: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.27 xhcievt: 0A40: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.28 xhcievt: 0A50: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.29 xhcievt: 0A60: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.29 xhcievt: 0A70: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.30 xhcievt: 0A80: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.31 xhcievt: 0A90: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.31 xhcievt: 0AA0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.32 xhcievt: 0AB0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.33 xhcievt: 0AC0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.33 xhcievt: 0AD0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.34 xhcievt: 0AE0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.34 xhcievt: 0AF0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.35 xhcievt: 0B00: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.36 xhcievt: 0B10: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.36 xhcievt: 0B20: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.37 xhcievt: 0B30: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.38 xhcievt: 0B40: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.38 xhcievt: 0B50: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.39 xhcievt: 0B60: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.40 xhcievt: 0B70: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.40 xhcievt: 0B80: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.41 xhcievt: 0B90: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:04.42 xhcievt: 0BA0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
[... 89 lines cut ...]
00:00:05.01 xhcidcbaa: 00D0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:05.02 xhcidcbaa: 00E0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:05.02 xhcidcbaa: 00F0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:05.03 xhcidcbaa: 0100: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:05.04 xhcidcbaa: 0110: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:05.04 xhcidcbaa: 0120: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:05.05 xhcidcbaa: 0130: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00
00:00:05.06 xhcidcbaa: 0140: 00 00 00 00 00 00 00 00-F5 EF CC DD 50 46 97 EA
00:00:05.06 xhcicap: Dumping 0x1C bytes starting at 0x600000000
[... cut ...]
00:00:05.11 xhciop: 0050: 40 00 60 00 00 00 00 00-20 00 00 00 00 00 00 00
00:00:05.11 xhciir0: Dumping 0x20 bytes starting at 0x600000220
00:00:05.12 xhciir0: 0220: 02 00 00 00 F4 01 00 00-01 00 00 00 00 00 00 00
00:00:05.13 xhciir0: 0230: C0 10 60 00 00 00 00 00-88 08 60 00 00 00 00 00
00:00:05.13 pcie: Bridge status is 0x10
00:00:05.14 pcie: Device status is 0x18
00:00:05.14 xhci: 2929 KB shared memory free
00:00:05.14 kernel: Compile time: Mar  8 2021 21:02:03
00:00:05.15 kernel: Plug in an USB flash drive!
```

rsta2, 2021-03-09:
Thanks for the log! For some still unknown reason the MSI interrupt from the xHCI is not triggered. I need more time and will come back, when I have an idea, what happens here.

dwhinham, 2021-03-09:
No problem at all Rene, take as much time as you need - I appreciate it, thankyou!

rsta2, 2021-03-09:
Thanks Dale. Not sure, but I guess there is some chance that the USB code, built from Circle Step43.3, runs successfully with the firmware downloaded in boot/, when Step43.2 is checked out. My guess is, that there were some changes in the firmware, which are incompatible with Circle.

Can you please try the sample/39-umsdplugging from Step43.3 with the older firmware from Step43.2 to verify this hypothesis? Thanks.

rsta2, 2021-03-09:
Thanks for testing! So I was wrong. The xHCI driver tries to enable a device slot by sending the related command for the internal USB 2.0 hub (and the USB flash drive, if attached), but the answer on this command in the event ring is not seen, because the PCIe MSI interrupt does not come through to the driver. So the command times out. The question is, why the MSI does not come through?

Thank you Dale, for the offer to lend me your RPi to get this working. I will write an email to you.

rsta2, 2021-03-10:
The issue should be fixed on the *develop* branch now.

@dwhinham Thank you for your help!

dwhinham, 2021-03-10:
I can confirm this appears to have fixed USB on my Pi 4!

Great work as always @rsta2 and thanks again, I think this can be closed. 🙂

closed 2021-03-10 by dwhinham