rsta2/circle #211: USB storage issues (Pi 3 A+/B/B+)

dwhinham, 2021-03-10:
Hi Rene,

As promised, here is my other USB issue (sorry!).

In my project we often load large (>200MB) files from SD card. Some users would like to move these files onto USB sticks for convenience (and potentially faster loading, especially on Pi 4 with USB 3.0), so I've been experimenting with USB file I/O.

On Pi 4, everything seems rock-solid, but on Pi 3, we have mixed results. There are often `Transaction failed` errors in the console or other errors such as `umsd: Cannot clear halt on endpoint 1`.

To try and narrow this down, I've written a simple test kernel ([source tree here](https://github.com/dwhinham/circle/tree/storage-test)) which reads a large file into memory and performs a SHA sum to verify it loaded correctly, a bit like our earlier SD card tests.

Here is a binary with large data file included: [Google Drive download](https://drive.google.com/file/d/1cdmewqtnH1DDxDIdZBTQxF-TDTJyBuAF/view?usp=sharing)

- Inside the archive is a `testfile.bin` - this file needs to be moved to a FAT32 USB stick.
- The test is based on Circle 43.3.
- There are two 64-bit kernels for Pi 3, `kernel8-nopnp.img` expects a USB stick to be inserted from boot at all times, `kernel8-pnp.img` will ask you to insert/remove the stick for each test cycle a little bit like `39-umsdplugging`.
- The kernel will try to read the file, compare it with a known SHA256 hash, and then write it back to the stick as `testfile2.bin`.
- The non-PnP kernel will repeat this forever; the PnP kernel will let you unplug and replace the stick.

The USB stick model doesn't seem to matter; I've had similar results with various unbranded sticks, but I've mainly being testing a Sandisk Ultra USB 3.0 64GB (possibly also known as a Sandisk Cruzer?).

### Pi 3 **A+**
#### PnP kernel (FAIL)
```
logger: Circle 43.3 started on Raspberry Pi 3 Model A+ (AArch64)
00:00:00.66 timer: SpeedFactor is 1.51
00:00:00.69 sdhost: emmc1: sdhost-bcm2835 loaded
00:00:02.20 emmc: Found a valid version 3.0x SD card
00:00:03.35 dwhci: No device connected to root port
00:00:03.36 kernel: Compile time: Mar 10 2021 22:14:02
00:00:03.36 kernel: Plug in an USB flash drive!
00:00:09.27 usbdev0-1: Device ven781-5581 found
00:00:09.28 usbdev0-1: Interface int8-6-50 found
00:00:09.28 usbdev0-1: Using device/interface int8-6-50
00:00:09.44 umsd: Capacity is 58656 MByte
00:00:09.45 dwroot: Device configured
00:00:09.45 kernel: Reading 265MB from USB:...
00:00:11.26 dwhci: Transaction failed (status 0x80)
00:00:11.27 umsd: CSW transfer failed
00:00:11.27 umsd: TryRead failed
00:00:11.27 dwhci: Transaction failed (status 0x8)
00:00:11.28 umsd: Cannot clear halt on endpoint 1
00:00:11.28 kernel: Failed to read /testfile.bin into memory
```

#### Non-PnP kernel (FAIL)
```
logger: Circle 43.3 started on Raspberry Pi 3 Model A+ (AArch64)
00:00:00.66 timer: SpeedFactor is 1.51
00:00:00.69 sdhost: emmc1: sdhost-bcm2835 loaded
00:00:02.20 emmc: Found a valid version 3.0x SD card
00:00:03.10 usbdev0-1: Device ven781-5581 found
00:00:03.10 usbdev0-1: Interface int8-6-50 found
00:00:03.11 usbdev0-1: Using device/interface int8-6-50
00:00:03.26 umsd: Capacity is 58656 MByte
00:00:03.27 dwroot: Device configured
00:00:03.28 kernel: Compile time: Mar 10 2021 22:51:10
00:00:03.28 kernel: Reading 265MB from USB:...
00:00:04.07 dwhci: Transaction failed (status 0xA2)
00:00:04.08 umsd: Data transfer failed
00:00:04.08 umsd: TryRead failed
00:00:04.08 dwhci: Transaction failed (status 0x8)
00:00:04.09 umsd: Cannot clear halt on endpoint 1
00:00:04.09 kernel: Failed to read /testfile.bin into memory
```

### Pi 3 **B**

#### PnP kernel (OK, but has errors in the log!)
```
logger: Circle 43.3 started on Raspberry Pi 3 Model B (AArch64)
[... 198 lines cut ...]
00:00:54.32 kernel: Performing SHA256 hash...
00:01:09.29 kernel: SHA256 sum for /testfile.bin: 7f8b25dd6fd03f90d7e903b7b07b07d9a758c41cefb7e656a258eec3d4836c78
00:01:09.30 kernel: SHA256 sum MATCH
00:01:10.70 kernel: Writing 265MB to USB:...
00:01:29.07 kernel: File written in 18.36 seconds
00:01:29.07 kernel: === TEST PASSED! (2/2) ===
00:01:29.08 kernel: Reading 265MB from USB:...
00:01:37.78 kernel: File read in 8.68 seconds
00:01:37.78 kernel: Performing SHA256 hash...
00:01:52.74 kernel: SHA256 sum for /testfile.bin: 7f8b25dd6fd03f90d7e903b7b07b07d9a758c41cefb7e656a258eec3d4836c78
00:01:52.75 kernel: SHA256 sum MATCH
00:01:54.36 kernel: Writing 265MB to USB:...
00:02:53.05 kernel: File written in 58.67 seconds
00:02:53.05 kernel: === TEST PASSED! (3/3) ===
...
```

So, to summarize:
- 3A+ has trouble initializing the stick with this test kernel in PnP and Non-PnP.
  * In my actual application, which has much more going on (multi core, timer interrupts, audio), the stick *IS* initialized and can be read for some time, but will usually fail with a similar error halfway through reading a second file.
- 3B complains of Transaction Failed with PnP kernel, even though the read data integrity is good. Seem OK in Non-PnP kernel.

I'm not sure how the test performs on a 3 B**+**, as unfortunately my 3 B+ was accidentaly damaged a few days ago, but I was wondering if you are able to reproduce any of the above to start with.

Thanks again for your time!

rsta2, 2021-03-11:
Thanks for the report and the valuable test kernel!

First I tried it on a RPi 3B+ and it behaves like the RPi 3B (a number of "Transaction failed (status 0x202)" messages, but transfers are OK). These messages come from polling the hub controller of the on-board hub(s) of the RPi 3B(+) for plug-and-play events and are not serious. They can occur and the hub driver is restarting the transfer silently, if an transaction fails with this status.

Now, because I do not have an RPi 3A+, I tested your test kernel on an RPi Zero W and it shows the same behavior like the RPi 3A+. What makes the difference here is the USB hub, which is integrated on the RPi 3B(+), but not on the RPi 3A+ and RPi Zero W. All these RPi models have the same USB host controller. I attached an external USB hub to the RPi Zero W and with the USB flash drive connected to this external hub, all transfers are OK.

Of course the aborted transfers are serious and I think, I can analyze this further using the RPi Zero W, which will take some time.

dwhinham, 2021-03-11:
Thankyou Rene, as always, take as much time as you need 🙂

I guessed that the differences would be because of the lack of the integrated hub on the Pi 3A+, so a good idea to to test on the Zero W, I didn't think of that!

Regardless, a Pi 3 A+ is on its way to you from a supplier in Germany and it should hopefully be with you in a few days - even if you manage to find a solution without it, I hope it might be useful to you; it's a nice small form factor.

Cheers! 🙂

rsta2, 2021-03-11:
Thank you very much, Dale!

rsta2, 2021-03-15:
Hi Dale, I corrected the hard errors in reset recovery and now your test kernel works better on the RPi 3A+. One may still get error messages, but in most cases the driver can recover from these conditions. Unfortunately this still does not work in all cases yet. It is probably necessary to totally suppress these error conditions in the DWHCI driver and I tried several things, but wasn't successful so far. Unfortunately there will probably not a quick fix for this issue.

rsta2, 2021-03-16:
Yes, I guess, it's a timing issue. But I was playing with some parameters in this direction without success. It's also lack of knowledge. The Linux driver, which I have analyzed to implement the Circle driver, was rather complex. I may have missed something.

rsta2, 2021-03-18:
Now I found a way to retry failed USB bulk transactions on a low level, so that the storage driver will not notice it. Your test kernel did run here one hour without any problem with a SanDisk Ultra Flair USB 3.0 32GB flash drive with this modification on a RPi 3A+. I hope, this fixes this issue for you too. Can you please test it. Thanks! The patch is on the *develop* branch.

dwhinham, 2021-03-20:
Hi Rene,

This is amazing - I think you've done it. I didn't even bother with the test kernel and I applied these new commits directly to my project. Thankyou so much!

USB mass storage now seems to be stable with my 3A+ and my 3B - I can load large files from USB stick reliably.
As you mentioned, I still get "Transaction failed" on the 3B, but the file loads successfully as expected. On the 3A+ the log is always clean!

I've decided to release a new version of my project with USB storage as an "experimental" feaature. That way we can get more feedback as I have quite a number of active users.

Already though this is looking very, very promising. Thankyou for the time you have spent looking into this!
Have a great weekend! 😃

rsta2, 2021-03-20:
Hi Dale, thank you, also for testing! I appreciate that you have created a release including these modifications. The feedback from your users will be very valuable. Have a great weekend too!

philtimmes, 2021-04-08:
Could you suppress (disable) plug&play polling during active transactions, and then re-enable them afterwards?
_EnterFileOpenCritical();
_ExitFileOpenCritical();
??

rsta2, 2021-04-08:
@philtimmes For what purpose should I do this?

rsta2, 2021-05-14:
This issue has been fixed in Circle 44. So this can be closed.

closed 2021-05-14 by rsta2