rsta2/circle #213: I multi-core

subbbbbaru, 2021-03-25:
Hi,

I am learning to create multi-core applications for Raspberry Pi 3.
I want to create application using circle with mutli-core support on Raspberry Pi 3 Model B.
Example:

Core 0 lights the LED (GPIO pin 18) for 500 milliseconds.
Core 3 lights the LED (GPIO pin 19) for 100 milliseconds. 
Core 1 and 2 are shutdown.

I would be happy if you may help me.
Thank you

rsta2, 2021-03-25:
Welcome. I attached a sample program, which does the requested tasks. It also initializes the screen and UART and writes some messages to the screen (by default). The files in the archive have to be extracted into the *app/multicoresimple/* directory in a cloned Circle project tree. You have to create a file *Config.mk* in the Circle project root with this contents, to configure Circle before build (remove the # for 64-bit mode):

```
#AARCH = 64
RASPPI = 3
DEFINE += -DARM_ALLOW_MULTI_CORE
```

[multicoresimple.zip](https://github.com/rsta2/circle/files/6203217/multicoresimple.zip)

subbbbbaru, 2021-03-25:
Thank you very much for your help.

closed 2021-03-25 by subbbbbaru