rsta2/circle #214: Build error

l-o-l, 2021-04-03:
I'm trying to build from a Pi4 and installed the Arch64 tool-chain. 

I get a bunch of errors when doing make clean on the lib folder:

/usr/local/gcc-arm/bin/aarch64-none-elf-g++: 3: /usr/local/gcc-arm/bin/aarch64-none-elf-g++: Syntax error: word unexpected (expecting ")")
/usr/local/gcc-arm/bin/aarch64-none-elf-g++: 1: /usr/local/gcc-arm/bin/aarch64-none-elf-g++: ELF�0d@@��@8: not found
/usr/local/gcc-arm/bin/aarch64-none-elf-g++: 2: /usr/local/gcc-arm/bin/aarch64-none-elf-g++: @!@@@@@0pp@p@@�n�n�v�vQ�vQ�/ІX�X�QX�Q���@�@DD�v�vQ�vP�td@o@oO@oO�%�%Q�tdR�td�v�vQ�vQhh/lib/ld-linux-aarch64.so.1GNUGNU0J��a���^�slԛW�: not found
/usr/local/gcc-arm/bin/aarch64-none-elf-g++: 3: /usr/local/gcc-arm/bin/aarch64-none-elf-g++: Syntax error: word unexpected (expecting ")")
  CLEAN  /home/pi/src/circle/sample/40-irqlatency

Any help would be greatly appreciated.

l-o-l, 2021-04-04:
Sorry, my mistake. I had assumed there were instructions to build from the PI itself.

closed 2021-04-04 by l-o-l

rsta2, 2021-04-04:
Circle can be build from Raspberry Pi OS (32-bit) itself. That does not work for all components, because some need support from the math library, which is not compatible there, but the main libraries should build. I haven't used 64-bit Linux on the RPi by myself so far.

These errors are strange. How it seems, the compiler is reading its own binaries as source files. That cannot work. You could try to disable dependency checking by setting th parameter `CHECK_DEPS` to 0 in *Rules.mk*. Then `make clean` should work at least and with a cleaned project you can try again with `make`.

l-o-l, 2021-04-04:
I found your comment in the big thread, did this:

#PREFIX	 ?= arm-none-eabi-
PREFIX ?= 

And now I can ./makeall clean. Sorry, I am completely new to pi/linux. The issue was that I had put the tool chain on my machine and the binaries are of course not for arm (I don't know why the error was the way it was).

It's currently ./makeall 'ing happily.

rsta2, 2021-04-04:
If you want to build for AArch64, you have to set `AARCH = 64` in Rules.mk. `RASPPI` sets the major version of the target RPi model. With `AARCH = 64` the toolchain prefix is `PREFIX64`. You can create a file *Config.mk* in the root directory of the Circle project with this contents:

```
AARCH = 64
RASPPI = 4        # or 3 for RPi 3 target
PREFIX64 = /usr/local/gcc-arm/bin/aarch64-none-elf-
```

You do not need to modify *Rules.mk* then.

l-o-l, 2021-04-04:
Thank you very much for helping. I'm still very much at the "just get something to blink" stage of setting things up, so will stick with the 32bit for now, but I'll try this when the time comes!