rsta2/circle #215: CM3+ EMMC Fatfs

philtimmes, 2021-04-03:
Have tried both SDHost and Emmc methods, can't seem to access EMMC module on CM3+, even though its booted from it.

rsta2, 2021-04-04:
Unfortunately the eMMC is not supported on the CM3+ by Circle. Actually it will only be supported on the CM4 in the next release. That it boots from the eMMC is not a contradiction. That is done completely by the boot code installed in the hardware and by the firmware.

philtimmes, 2021-04-08:
Is there any indication what the problem is with the EMMC on the CM3?

rsta2, 2021-04-08:
The eMMC needs a slightly different command handling, compared to a normal SD card. This can be enabled with the system option `USE_EMBEDDED_MMC_CM4`, but then the EMMC2 controller of the CM4 will be used too, so it will not work. There may also be issues with the SD card power supply. It's simply not supported on the CM3(+).

DLehenbauer, 2021-04-26:
@rsta2 - Is EMCC/SDHost with RPi3 expected to work, or is the answer the same as CM3?

My RPi3 is failing in CardReset().  Same code works on the RPi0.  (If there's interest, I can collect logs and more info.)

Also, fantastic project!

rsta2, 2021-04-27:
@DLehenbauer Thanks for appreciating this project! The SD card drivers in *addon/SDCard* should work via both supported interfaces (EMMC, SDHOST) on the RPi 3 (B, B+, A+).

You can define `EMMC_DEBUG2` [here](https://github.com/rsta2/circle/blob/master/addon/SDCard/emmc.cpp#L58) and set `SDHOST_DEBUG` to 1 [here](https://github.com/rsta2/circle/blob/master/addon/SDCard/sdhost.cpp#L44) to get more debug output. You will probably need a serial interface to capture all this output.

DLehenbauer, 2021-04-28:
Oh!  I forgot that SDCard was an /addon/ and forgot to recompile it when switching platforms.  It works great on RPi3.  Thx!

rsta2, 2021-04-28:
No problem, better that it works now. Thanks for info.

closed 2021-04-28 by rsta2

hires, 2021-09-26:
Based on your understanding of the Pi hardware, what would be needed to add support for eMMC on the CM3+? Linux obviously can access it, so I assume there is a way. I am attempting to make a bootloader that can write directly to the CM3+ eMMC without the need for the usbboot program which unfortunately mounts as several disks and is confusing for end-users. I've been meaning to use Circle for a project and kind of sad that this isn't possible.

Willing to hack on it if there is hope.

rsta2, 2021-09-26:
Extending the EMMC driver for new hardware was mostly a try-and-error process in the past. You have to modify several things and see if it works. You can only read the [EMMC driver](https://github.com/rsta2/circle/blob/master/addon/SDCard/emmc.cpp) carefully, find the `#ifdef` statements and see, what may affect eMMC operation on the CM3+. You should enable the `EMMC_DEBUG2` define in line 59, because this produces much more debugging output. The reason why I haven't done this before is that I don't have a CM3+ here. I cannot test it.

hires, 2021-09-26:
Okay thanks I will work on it.

philtimmes, 2021-09-26:
The issue with the emmc on the cm3 and cm3+ is the voltage testing. The cm3
does not use the internal voltage regulator for the emmc, it is locked at
1.8V. and it fails to pass the voltage negotiation portion. Additionally,
the emmc MUST be reset, but the reset process needs tweeking. I had it
working on my builds, but abandoned because gpio limited to 15.8MHz on cm3+.

hires, 2021-09-28:
philtimmes: What do you mean by "gpio limited to 15.8MHz on cm3+"?

philtimmes, 2021-09-28:
I mean that any GPIO cannot be sampled, or have its state changed any
faster than 15,800,004 times per second. This is unrelated to the EMMC
issue, but is why we abandoned the CM3+ as a platform for our product.