rsta2/circle #217: CCPUThrottle triggers rebooting if enabling multi-core?

sjtugjy, 2021-04-12:
I make a test based on the example of 26-cpustress as follows.

I first comment out all the statements like m_pScreen->SetPixel and then make one function (CTemperatureTask::Run) simply execute the loop below.

void CTemperatureTask::Run (void) {
    volatile int test;
    while (1) {
    CCPUThrottle::Get ()->SetSpeed (CPUSpeedMaximum, TRUE);
    for (test = 0; test < 100000000; ++test) ;
    CCPUThrottle::Get ()->SetSpeed (CPUSpeedLow, TRUE);
    for (test = 0; test < 100000000; ++test) ;
    }
}

Everything seems OK without defining ARM_ALLOW_MULTI_CORE.
However, if I define it in sysconfig.h, the same test triggers system rebooting.

sjtugjy, 2021-04-12:
The vanilla example also cannot run after enabling smp.

rsta2, 2021-04-12:
I cannot reproduce this here. Please attach the complete test program source and tell me, on which Raspberry Pi model, in which mode (32 or 64 bit) and after which runtime of the program, you have noticed the reboot.

sjtugjy, 2021-04-12:
[Uploading temptask.cpp…]()
[Uploading mandelbrot.cpp…]()

In the latest master, I just modify these two files.
The hardware is 3b+ and 64-bit.

sjtugjy, 2021-04-12:
/home/tmac/rasp-boot/circle/sample/26-cpustress/temptask.cpp

sjtugjy, 2021-04-12:
[temptask.txt](https://github.com/rsta2/circle/files/6296213/temptask.txt)
[mandelbrot.txt](https://github.com/rsta2/circle/files/6296214/mandelbrot.txt)

In the latest master, I just modify these two files.
The hardware is 3b+ and 64-bit. The firmware is also from the master.

rsta2, 2021-04-12:
Thanks. I still cannot reproduce this problem. I tried exactly the same configuration like yours and it was running without reboot for 8 minutes now.

Have you cleaned the Circle project after enabling `ARM_ALLOW_MULTI_CORE` using `makeall clean` and the sample source using `make clean`? Otherwise there may be old object files, which were compiled without `ARM_ALLOW_MULTI_CORE`. This will not work.

sjtugjy, 2021-04-12:
Yes, I have cleaned them all.

Is that possible due to the power supply? I use the MicroUSB one with 5V/2.5A.

sjtugjy, 2021-04-12:
I guess it is due to the power problem. After I use another charge (3A), it works.

sjtugjy, 2021-04-13:
It works now.

closed 2021-04-13 by sjtugjy

rsta2, 2021-04-14:
OK, thanks for info.

sjtugjy, 2021-04-14:
You are welcome. Actually, I had planned to add a note in README of that example but failed to make a pull request.

rsta2, 2021-04-15:
A note regarding the power supply? It's a bit strange, that a 2.5A power supply is not enough. The one I'm using here provides 2.0A, if I can believe the info on it, and it is working well with that sample.

sjtugjy, 2021-04-15:
Hmm, it's weird.