rsta2/circle #221: USBSimple and USBKeyboard is not working for my Raspberry Pi 4

luisllanos9, 2021-04-27:
Hello, 
I got a problem with the samples 05-usbsimple and  08-usbkeyboard, they are not working for my raspberry pi 4 Model B.
I don't have to many information to solve this problem, I just turn on the raspberry pi and this is the message that appears:

logger: Circle 43.3 started on Raspberry Pi 4 Model B (AArch64)
00:00:00.66 timer: SpeedFactor is 1.51
00:00:00.73 pcie: Link up, 5.0 Gbps x1
00:00:04.04 xhcidev: Cannot enable slot (-2)
00:00:04.04 kernel: Compile time: Apr 27 2021 11:07:18
00:00:04.04 kernel: USB hub not found

Can someone solved for Raspberry Pi 4 Model B? :)

rsta2, 2021-04-27:
Can you please try the Circle version from the *develop* branch (`git checkout develop`)? There was an issue with some RPi 4B models, which has been fixed in the meantime.

rsta2, 2021-05-14:
This issue should be fixed in Circle 44. So this can be closed.

closed 2021-05-14 by rsta2