rsta2/circle #224: USB Keyboard not working with REALTIME

XORElectronics, 2021-05-03:
Hello,

I am trying to use a USB keyboard in combination with the REALTIME setting from my application.
Getting completely back to a proof of concept and using the sample project 08-USB Keyboard. Latest stable Circle release, GCC10.2.0.   
I use the Raw mode, but the normal mode also doesn't work as it should. 
Basically, the Keyboard gets detected well (also when I remove it it detects that it's removed), but then the log-rotation is not 'running'. 
(It does look to me that it doesn't enter the keyboard handlers except for the one to remove)
The flag USE_USB_SOF_INTR doesn't make a difference here except for the message on the screen that it is recommended.

Using 43.3 but tried also todays develope branch. Raspberry 1 Zero, compiling with -O4 

All works as it should when REALTIME is disabled. Maybe I understood it wrong, but I thought that with the 43.x releases of circle this should work together.(And I hope so)
Thanks!

rsta2, 2021-05-03:
Yes, the sample/08-usbkeyboard does unfortunately not work with the REALTIME option, but not because the keyboard is not working. The problem is, that the KeyPressedHandler() or KeyStatusHandlerRaw() are called at IRQ_LEVEL and these handlers try to write directly to the screen, which is forbidden with REALTIME enabled to keep the IRQ latency low.

If you want to use the REALTIME option, you have to implement some buffering to decouple the screen output from the IRQ handler. There is a slightly modified sample/08-usbkeyboard attached, which uses the class CWriteBufferDevice to do this. You may implement this buffering on the keyboard side too, write received characters into a ring buffer and read them on TASK_LEVEL to display them.

[08-usbkeyboard.zip](https://github.com/rsta2/circle/files/6417247/08-usbkeyboard.zip)

XORElectronics, 2021-05-03:
Oh ok i see! Didn't think about that but that makes of course sense. 
Thanks!

closed 2021-05-03 by XORElectronics