rsta2/circle #226: Suggestion - Add CMake capability

Memotech-Bill, 2021-05-10:
It would be nice if Circle had the CMakeLists.txt files needed to turn Circle into a CMake INTERFACE library similar to the SDK for the RPi Pico.

This would make it easier to have projects that will build for either Pico or bare-metal RPi.

rsta2, 2021-05-11:
I have to do some reading, when I have some time left, because I did not use CMake so far. Currently I think, the Circle build system is good. It does work with toolchain, make, shell only, is easy to setup and configure and checks file dependencies. So CMake could only be another option and I have to check, if the additional effort is worth doing.

Memotech-Bill, 2021-05-11:
One further advantage of CMake is that it does out of source tree builds. This means that it would be easy to have different build folders for each Raspberry Pi type (1-4), rather than having to do a "make clean" and a full rebuild when changing type.

Memotech-Bill, 2021-05-12:
For information, I have now got CMake to compile enough of Circle for my purposes. My approach is slightly different in that the CMake definitions are outside the Circle source tree, but may still be useful. In my main CMakeLists.txt file I have:

`
  set(CMAKE_CXX_STANDARD 14)

  project(memu C CXX ASM)

  set(CIRCLE_DIR ${CMAKE_CURRENT_LIST_DIR}/../circle)

  set(CMAKE_LINKER ld)
  set(CMAKE_CXX_LINK_EXECUTABLE
    "<CMAKE_LINKER> -o <TARGET> <LINK_FLAGS> <OBJECTS> --start-group <LINK_LIBRARIES> --end-group")

  if (${RPI} EQUAL 1)
    set(ARCH -DAARCH=32 -mcpu=arm1176jzf-s -marm -mfpu=vfp -mfloat-abi=hard)
    add_compile_definitions(RASPPI=1)
    set(KERNEL kernel.img)
  elseif (${RPI} EQUAL 2)
    set(ARCH -DAARCH=32 -mcpu=cortex-a7 -marm -mfpu=neon-vfpv4 -mfloat-abi=hard)
    add_compile_definitions(RASPPI=2)
    set(KERNEL kernel7.img)
  elseif (${RPI} EQUAL 3)
    set(ARCH -DAARCH=32 -mcpu=cortex-a53 -marm -mfpu=neon-fp-armv8 -mfloat-abi=hard)
    add_compile_definitions(RASPPI=3)
    set(KERNEL kernel8-32.img)
  elseif (${RPI} EQUAL 4)
    set(ARCH -DAARCH=32 -mcpu=cortex-a72 -marm -mfpu=neon-fp-armv8 -mfloat-abi=hard)
    add_compile_definitions(RASPPI=4)
    set(KERNEL kernel7l.img)
  endif()

  add_compile_definitions(__circle__)
  add_compile_options(${ARCH})
  add_compile_options($<$<COMPILE_LANGUAGE:CXX>:-fno-exceptions>)
  add_compile_options($<$<COMPILE_LANGUAGE:CXX>:-fno-rtti>)
  add_compile_options($<$<COMPILE_LANGUAGE:CXX>:-nostdinc++>)
  add_compile_options(-O2)
  add_compile_options(-Wall -fsigned-char -ffreestanding)
  add_compile_options($<$<COMPILE_LANGUAGE:CXX>:-std=c++14>)
  add_compile_options($<$<COMPILE_LANGUAGE:CXX>:-Wno-aligned-new>)
  add_link_options(--section-start=.init=0x8000)
  add_link_options(-T ${CIRCLE_DIR}/circle.ld)
  add_compile_definitions(
    STDLIB_SUPPORT=1
    __VCCOREVER__=0x04000000
    )
  add_compile_options(
    -U__unix__
    -U__linux__
    )

  execute_process(COMMAND ${CMAKE_C_COMPILER} ${ARCH} -print-file-name=libgcc.a
                OUTPUT_VARIABLE LIBGCC_PATH
                OUTPUT_STRIP_TRAILING_WHITESPACE)

  execute_process(COMMAND ${CMAKE_C_COMPILER} ${ARCH} -print-file-name=libm.a
                OUTPUT_VARIABLE LIBM_PATH
                OUTPUT_STRIP_TRAILING_WHITESPACE)
              
  add_executable(memu-circle)

  target_compile_definitions(memu-circle PUBLIC
    -DAARCH=32
    -D__circle__
    )

  include_directories(
    ${CIRCLE_DIR}/include
    ${CIRCLE_DIR}/addon
    ${CMAKE_CURRENT_LIST_DIR}/circle/include
[... 239 lines cut ...]
  ${CIRCLE_DIR}/lib/sched/synchronizationevent.cpp
  )

add_library(circle_input STATIC)

target_sources(circle_input PRIVATE
  ${CIRCLE_DIR}/lib/input/console.cpp
  ${CIRCLE_DIR}/lib/input/keyboardbehaviour.cpp
  ${CIRCLE_DIR}/lib/input/keyboardbuffer.cpp
  ${CIRCLE_DIR}/lib/input/keymap.cpp
  ${CIRCLE_DIR}/lib/input/linediscipline.cpp
  ${CIRCLE_DIR}/lib/input/mousebehaviour.cpp
  ${CIRCLE_DIR}/lib/input/mouse.cpp
  ${CIRCLE_DIR}/lib/input/touchscreen.cpp
  )

add_library(circle_fs)

target_sources(circle_fs PRIVATE
  ${CIRCLE_DIR}/lib/fs/partition.cpp
  ${CIRCLE_DIR}/lib/fs/partitionmanager.cpp
  )
`

For an in-source-tree implementation, this latter file would be split into multiple CMaleLists.txt files, one per folder, and the source files specified as ${CMAKE_CURRENT_LIST_DIR}/filename.

rsta2, 2021-05-12:
Thanks for showing, how it may work. But as I told, CMake can only be another option for a Circle build system. There are many projects out there, which use the current build system and all them would have to be modified, if I would come with a new system now. This will not happen. I also do not see that much advantages, over the current build system. That one can have multiple object versions is a nice, but honestly Circle builds quick enough to rebuild it, if required, because it's small. I think, there are more people, which are familiar with make than with CMake (including me). I do not see, CMake in Circle in the near future. Changing the build system is a big step.

kevinboone, 2021-11-18:
I'd like to point out that the use of CMake is a _disincentive_ to many people working on the Pi Pico. I'm aware that some people like it and, of course, that's fine. It would be a sad world if we were all the same. However, the problem with the Pico is that it's almost impossible to work without using CMake. CMake is fine for some things, but it's highly opinionated. Personally, I think the way that Circle builds is fine as it is. Just my $0.02, of course.

rsta2, 2021-11-18:
Thanks @kevinboone. Good to hear your opinion. Recently I worked with the RPI Pico and I think, using CMake with it is OK. But this does not mean, that it is necessary to use CMake for Circle too, when you want to share files between the projects. So I think that the Circle build system can remain and this issue can be closed soon.

Memotech-Bill, 2021-11-18:
I never intended to suggest that CMake should be the only build option for Circle, only that it would be nice to have a CMake file available as an alternate build option. I do recognise the extra work involved in maintaining multiple build options.

I wrote my own CMake file to build what I needed from Circle.

As far as I am concerned this issue may be closed.

rsta2, 2021-11-18:
@Memotech-Bill Thanks for clarifying your position. Yes, there are many Makefiles in Circle and maintaining another build system would require more time, which can probably be used better for other things. And if you have solved this problem for yourself, I will close this issue.

closed 2021-11-18 by rsta2