rsta2/circle #227: measure the execution time of my code

subbbbbaru, 2021-05-13:
Hi,

I want to measure the execution time(milliseconds) of my code.  How can i do this?

I would be happy if you may help me.
Thank you

rsta2, 2021-05-13:
If you have the `CTimer` class instantiated in your application, you can call:

```
#include <circle/timer.h>

unsigned nStartTicks = CTimer::GetClockTicks();

// your code here

unsigned nEndTicks = CTimer::GetClockTicks();
unsigned nDurationMs = (nEndTicks - nStartTicks) / 1000;
```

`GetClockTicks()` returns the current value of a free running 1 MHz counter. `nDurationMs` will contain the execution time of your code in milliseconds.

subbbbbaru, 2021-05-13:
Thank you very much for your help.

closed 2021-05-13 by subbbbbaru