rsta2/circle #229: UDP Send and Receive issue

giuseppe-avon, 2021-05-31:
I am using the latest Circle library on my RPi 4. 
Using an UDP socket I cannot manage to send and receive at the same time using the socket. To receive I need to Bind and then Receive, to send I need to connect and Send. Both the calls (connect+bind) are not possible. I am using the manual terminology, even if it is uncanny to me use connection-oriented terms in its opposite context.
Moreover, A Yield() call to the scheduler is needed to have the pending bytes sent. Sorry if I post here as an issue, which is for sure an usage issue more than an underlying implementation one.
RTFM is apparently not working, my code is inspired to NTP client internals.
What is the right way to allow an UDP socket to be able to send and receive?
Thanks in advance.

rsta2, 2021-05-31:
Using `Bind()` and `Connect()` on the same UDP `CSocket` instance should work. If you use `Connect()`, you can only receive UDP packets from the peer address, which you have specified in this call.

If you want to receive UDP packets from different senders, you can use `Bind()` only and `SendTo()` and `ReceiveFrom()` then instead of `Send()` and `Receive()`. Perhaps this is the way to go for you.

Please note that you have to call `SetOptionBroadcast (TRUE)` on the socket, if you want to send or receive any broadcast packets.

`CScheduler::Yield()` must be called from time to time, so that the network task can run in the background.

giuseppe-avon, 2021-05-31:
Thanks for your support. It works and seems more clear to me and now.
One last thing, which still stays in this TCP/IP networking argument: is there a way with the MSG_DONTWAIT to know if the send has occurred or failed (tcp)?

rsta2, 2021-06-01:
You are welcome. With TCP `CSocket::Send()` returns an error, if the transfer fails somewhere in the stream. This reaction can be delayed and it can't be told, at which position in the TCP stream the failure occurred.

giuseppe-avon, 2021-06-03:
Thanks again for your support.. Everything now is clear to me.

closed 2021-06-03 by giuseppe-avon