rsta2/circle #234: problems with USB on CM4 with 44.1

incanus, 2021-06-16:
I cannot get any of the USB examples to work on the CM4. Using the simplest (`05-usbsimple`), the full boot screen message I get is:

```
logger: Circle 44.1 started on Compute Module 4 (AArch64)
00:00:00.66 timer: SpeedFactor is 1.51
00:00:00.76 pcie: Link down
00:00:00.76 pcie: Cannot initialize PCIe bridge (-1)
00:00:00.76 xhci: Cannot initialize PCIe host bridge
```

And then there is no USB responsiveness or dump. 

Some things I have ensured and/or tried: 

- `otg_mode=1` in `config.txt`
- Micro USB is unplugged after flashing
- `usbpowerdelay=2000` in `config.txt` per https://github.com/rsta2/circle/issues/117#issuecomment-613329179
- unplugging/replugging USB keyboard (official Pi keyboard; only USB device attached

incanus, 2021-06-16:
While moving on and exploring the `USE_EMBEDDED_MMC_CM4`, I accidentally discovered `USE_XHCI_INTERNAL` in `include/circle/sysconfig.h`. Enabling that (of course) solves the problem. 

I wish that were better documented — this is my first time in that file. Can you suggest a place I might add it and I will submit a pull request with improvements around use of the CM4? 

At very least, this issue will remain in search results for others.

rsta2, 2021-06-17:
I have added a notice about *include/circle/sysconfig.h* to the main README file on the *develop* branch. Thanks for pointing on this! If you have more info on using the CM4 you could add a file *doc/compute-module.txt*.

incanus, 2021-06-21:
Thank you @rsta2! I don't have anything else re: CM4 to add right now (the only other requirement is `sounddev=sndhdmi`, which is well documented in the audio examples) so I'll close here but keep contributions in mind!

closed 2021-06-21 by incanus