rsta2/circle #235: Multiple serial devices for UART communication

tbobrow1, 2021-06-17:
Is it possible to initialize multiple CSerial devices so that I can use multiple UART communication lines plus the device serial monitor? I would like to have a serial monitor for displaying logs (nDev=0), and communication to two different arduinos (ndev=3; GPIO4/5 & nDev=4; GPIO8/9). Currently, when I try creating more than one serial device object, my kernel appears to be hanging while initializing. I am using an RPI4.

Thanks for putting this library together. It is awesome!

rsta2, 2021-06-17:
Yes, this should work on the RPi 4. You have to ensure, that for serial devices, which use an interrupt (IRQ or FIQ), you must specify the same configuration, that means you cannot use the IRQ for one serial device and the FIQ for another. All serial devices, which use an interrupt, must share the same kind of interrupt. But you can use a non-interrupt serial device (e.g. for the serial log) and two serial devices, which use the FIQ for example.

Attached there is a test program, which demonstrates this with three serial devices (nDevice=0, 4 and 5 here). I used it to test the serial support on the RPi 4 over a year ago. I hope, it helps you to get it running, but was not able to test it again now that quick. If it still not works for you, I would have to test it again. Please note that the test program automatically reboots after two minutes.

Thanks for appreciating Circle!

[serialtest-rpi4.zip](https://github.com/rsta2/circle/files/6672518/serialtest-rpi4.zip)

tbobrow1, 2021-06-19:
This is running perfectly on my device. Thanks for sharing the test code!

A slightly related question for you: do you have any recommendations for how I can utilize different configurations for the data/parity/stop bits? I am looking to utilize serial comms with 8 bit data, an even parity bit, and 1 stop bit.

Thanks!

rsta2, 2021-06-20:
The only supported serial data format is currently 8 data bits, no parity, 1 stop bit. I will add a method to select other formats. This will take some time.

rsta2, 2021-06-21:
`CSerialDevice` supports setting the line parameters on the *develop* branch now. You can call `m_Serial.Initialize (115200, 8, 1, CSerialDevice::ParityEven)` to select the requested settings.

tbobrow1, 2021-06-21:
This is perfect. Thanks very much for implementing, and for such a quick turnaround! I will go ahead and close out the issue. Best wishes to you.

closed 2021-06-21 by tbobrow1