rsta2/circle #238: CCharGenerator::GetCharWidth returning zero

giuseppe-avon, 2021-06-29:
During CScreen::Initialize() function, what are the conditions that could lead GetCharWidth() function returning a zero instead of the real character width size? Thanks.

giuseppe-avon, 2021-06-29:
To be more specific
CScreenDevice::ClearLineEnd (this=0x22f308 <piBridge+3592>) at screen.cpp:558
558             for (unsigned nPosX = m_nCursorX; nPosX < m_nWidth; nPosX += m_CharGen.GetCharWidth ())

m_CharGen.GetCharWidth() is always 0 and this for loop is never ending. This happens when I link my own library, which in turn calls the CStdLibApp from the Circle-StdLib.

rsta2, 2021-06-30:
`CCharGenerator::GetCharWidth()` can only return zero, when the constructor `CCharGenerator::CCharGenerator()` was not called properly (yet) or the `this` pointer is wrong.

rsta2, 2021-09-17:
I think, this can be closed.

closed 2021-09-17 by rsta2