rsta2/circle #239: CCPUThrottle doesn't work on Rpi4

miweber67, 2021-07-03:
CPU speed adjustments don't appear to work on the Raspberry Pi 4. I have a 1GB RPi4; built and ran the current master branch da58255 sample 26-cpustress. Temps climb to nearly 80C with zero change in the time required to calculate the full-view mandelbrot set with four cores active. With one core active, it takes too long for the temp to climb to a point where CCPUThrottle should be doing something. I added some logging to temptask.cpp; the logging indicates that CCPUThrottle believes it is adjusting the speed and reports a lower speed, but the actual speed of calculations does not slow down and the temperature continues to rise.

rsta2, 2021-07-04:
Unfortunately I cannot reproduce this problem. I tried the sample/26-cpustress with 4 cores on a 2 GB RPi 4B with the firmware downloaded in the boot/ subdirectory, the current stable firmware from May 27 2021 and the most recent firmware from the master branch from [here](https://github.com/raspberrypi/firmware/tree/master/boot). All tests showed the SoC temperature rising to 60 degrees Celsius and remaining there. Which firmware version are you using?

miweber67, 2021-07-05:
Thanks for your reply!

I tried a number of things. First, the makefile in the circle/boot folder default to pinning the firmware version 5985247fb...  from Feb 25 2021, so that is the version I was originally using. I commented that line out and used the 'master' firmware branch, but the behavior persisted. I also tried using the raspberrypi/firmware directory stock from the raspberrypi project instead of using the circle/boot folder contents, but that also did not work. I tried it both with and without using "fast=true" in the cmdline.txt. I tried with and without "socmaxtemp=45" in my cmdline.txt but it made no difference.

The only changes 'git status' shows in my build tree are the enabling of multi-core operation and the change to 'master' for the RPi firmware download in /boot.

I have attached the boot folder from my last attempt in case it's useful in debugging. I am happy to try more things or provide more details if that would help, but at this point, I am not sure what to look at next.

[boot_26.zip](https://github.com/rsta2/circle/files/6765231/boot_26.zip)

The upper-left Mandelbrot image takes 17 seconds to generate on my RPi 4B. Not sure what the clock speed is, since the results are the same with "fast=true" present or absent from cmdline.txt.

[image]

rsta2, 2021-07-06:
Thanks for providing the *boot_26.zip* archive. I copied all files to a SD card and the program runs here as expected. The SoC temperature remains at around 60 degrees Celsius. Even with the setting `socmaxtemp=45` in *cmdline.txt* the temperature cannot be hold down to 45 degrees Celsius, because even at the minimum CPU clock frequency of 600 MHz it gets hotter.

This is caused by the way, the class `CCPUThrottle` works. It simply measures the current SoC temperature and when it exceeds the `socmaxtemp=` setting, the CPU clock frequency will be to the minimum CPU clock frequency (600 MHz for the RPi 4) in the hope, that the temperature will drop. But if the temperature will not drop below the `socmaxtemp=` mark, the CPU clock will remain at the minimum CPU clock frequency all the time. That is all, what `CCPUThrottle` can do.

Is your RPi 4 enclosed in a case? I do not use a case here and 60 degrees Celsius is the temperature which can be enforced. This can vary a bit, because of different air temperatures around. Inside a case the RPi 4 can get very hot without further cooling actions.

miweber67, 2021-07-06:
Thanks for testing my image!

I am curious if you observe a difference in the time it takes to complete the mandelbrot tasks when the clock is reported to be 1500MHz vs. when it is reported to be 600MHz. On my pi, I see no difference, which would suggest to me that CCPUThrottle is not able to effect a change in the actual clock speed either up or down. The overall mandelbrot set (core 0 in multi-core mode) takes 17 seconds (by external stopwatch) to calculate regardless of the reported speed.

I installed the Pi in a box and let the temp rise to 87C to see if the built-in firmware would produce an observable effect under those conditions, and it did not (task time stayed the same, temp did not fall), which suggests to me that the clock speed is already as low as it can go and we are never actually getting it up to 1500. If the clock were actually 1500 and the temp was 87C I would have expected the Pi's built-in GPU-based firmware to do something about it, which I would have expected to show up as an increased time for the mandelbrot task.

My assumption is that the mandelbrot task should take measurably different time to complete at 1500MHz vs. 600MHz, which seems reasonable.

I am curious what you think of all of that? Thank you again for your help!

rsta2, 2021-07-06:
The top-left image takes about 4 seconds here, when the SoC temperature is below the `socmaxtemp=` value and about 9 seconds above. There is a visible speed change. The firmware should show one of these [warning icons](https://www.raspberrypi.org/documentation/configuration/warning-icons.md), when near or over 80 degrees Celsius.

miweber67, 2021-07-06:
OK! Mystery solved! I had a power supply issue. I thought I was using the official RPi power supply, but after tracing the mess of cables on my desk, discovered that I was using an older USB power adapter. I swapped cables to an actual RPi power supply, and suddenly things work as expected.

I did notice a lightning bolt in the upper-right-hand corner of the screen after you pointed out the warning icons. It looks like if you have the lightning bolt, you are automatically throttled to 600MHz.

Thanks again for your work and your help! Closing this issue.

closed 2021-07-06 by miweber67