rsta2/circle #240: USB keyboard double stroke when typing fast.

turco32, 2021-07-19:
Hello,

i am using a project someone made using circle: https://github.com/michaelengel/crosstalk
I thought id post here in advanced it is out of the developers hands. When i am typing at a normal speed, i press one key right after the other very quickly. Usually on something like linux or whatnot each key press will register as one keystroke. for some reason, using the above bare metal OS, the keys double or sometimes triple tap. Here is my current hardware setup.
Raspberry pi model: Pi Zero W
Keyboard/Mouse setup: Logitech K235 wireless keyboard/mouse combo
PSU: coming in at a steady 1.2V

If this isn't right of me to ask this here, i apologize.

rsta2, 2021-07-20:
This crosstalk project uses a relatively old version of Circle. There were a lot of changes and improvements to the USB stack in Circle since this version. There is some chance, that your problem will be fixed with the current Circle version 44.1. I tried the Circle *sample/08-usbkeyboard* with two different wired keyboards and the problem does not arise there, even when typing very fast. 

On the other hand there are some USB devices, which do not work reliable with Circle. It is difficult to fix such issues, because there are so many different USB devices out there and of course I cannot buy each device for testing.

turco32, 2021-07-20:
oh ok. The project owner did talk about updating the Circle version in an update, so ill wait out for that and see if it fixes the the issue. sorry to bother you!

closed 2021-07-20 by turco32