rsta2/circle #241: No way to read data from SPI slave vs1053 Audio Decoder

pelizz, 2021-08-01:
II try to interact with vs1053 AUDIO decoder (MP3/AAC) using Circle SPImaster support.
The SPI loopback test works, but when i connect VS1053  and i try to write/read register (for example VOLUME) i got always zero.
Here the connection [schema](https://3.bp.blogspot.com/-Ybq9lh0Vsk4/WeD_cLJ0I_I/AAAAAAAABMk/QC2XRRhxn2s6ENwK7FeAActRFQAc-yYagCLcBGAs/s1600/PinConnections.png)

...
for (int i = 0; i < 10; i++)
{
	writeRegister(SCI_VOL, (u8)i);
	CTimer::Get()->MsDelay(50);
	uint16_t result = readRegister(SCI_VOL);
}

inline uint16_t readRegister(uint8_t _reg)
{
	uint16_t result;
	u8 TxData[2];
	u8 RxBuffer[2] = {0xFF, 0XFF};

	control_mode_on();
	CTimer::Get()->usDelay(3);

	TxData[0] = (u8)3;
	TxData[1] = _reg;
	SPI.WriteRead(SPI_CHIP_SELECT, TxData, RxBuffer, 2);

	result = RxBuffer[0] << 8;
	result |= RxBuffer[1];
	CLogger::Get()->Write(FromVS1053driver, LogWarning, "4 WR:%d result:%d",RxBuffer[0],result);

	await_data_request(); // Wait for DREQ to be HIGH again
	control_mode_off();
	return result;
}

inline void writeRegister(uint8_t _reg, uint16_t _value)
{
	control_mode_on();

	u8 TxData[4];

	TxData[0] = (u8)2;
	TxData[1] = (u8)_reg;
	TxData[2] = (u8)_value >> 8;
	TxData[3] = (u8)_value & 0xff;
	SPI.WriteRead(SPI_CHIP_SELECT, TxData, 0, 4);

	CTimer::Get()->usDelay(1);

	await_data_request();
	control_mode_off();
}

rsta2, 2021-08-01:
I think your `readRegister()` function is not OK. In a SPI transfer data is sent (via MOSI) and received (via MISO) simultaneous by the master. According to the [VS1053 datasheet](https://cdn-shop.adafruit.com/datasheets/vs1053.pdf) (section 7.4.1) the VS1053 sends zero bits, while the instruction code and register address is received. I guess these zero values, you are receiving all the time are composed of these bits. Please try the following instead (untested):

```
inline uint16_t readRegister(uint8_t _reg)
{
	uint16_t result;
	u8 TxData[4];
	u8 RxBuffer[4] = {0xFF, 0XFF, 0xFF, 0xFF};

	control_mode_on();
	CTimer::Get()->usDelay(3);

	TxData[0] = (u8)3;
	TxData[1] = _reg;
	SPI.WriteRead(SPI_CHIP_SELECT, TxData, RxBuffer, 4);

	result = RxBuffer[2] << 8;
	result |= RxBuffer[3];

	await_data_request(); // Wait for DREQ to be HIGH again
	control_mode_off();
	return result;
}
```

You can also try to read the registers 0 and 1 first, without writing them before. According to section 9.6 of the datasheet these registers contain a non-zero value after reset and by reading them, you can try, if your read function works.

pelizz, 2021-08-02:
Hi...
it works ...thanks a lot.

I have also problems in the reset sequence ; the correct one is:

SPI.Initialize();
dreq_pin.SetMode(GPIOModeInput);
cs_pin.SetMode(GPIOModeOutput);
dcs_pin.SetMode(GPIOModeOutput);
xrst_pin.SetMode(GPIOModeOutput);

xrst_pin.Write(LOW);
dcs_pin.Write(HIGH); // Start HIGH for SCI en SDI
cs_pin.Write(HIGH);
CTimer::Get()->MsDelay(500);

xrst_pin.Write(HIGH);
writeRegister(SCI_VOL, 0xffff); // VOL
CTimer::Get()->MsDelay(500);

I have ported a driver for VS1053/1003 from ESP32 Environment to
Circle...and now it seems ok....
I will close the issue on Circle.

Again thank you.

Bye,
Maurizio

closed 2021-08-02 by pelizz