rsta2/circle #243: No screen output

xlar54, 2021-08-10:
Hey, could you help?  The following code just shows a blank screen with no text.  Obviously Im doing something wrong, but not sure what.

`
//
// kernel.cpp
//
// Circle - A C++ bare metal environment for Raspberry Pi

#include "kernel.h"
#include <circle/gpiopin.h>
#include <circle/timer.h>
#include <circle/screen.h>
#include <circle/koptions.h>

CScreenDevice m_Screen = CScreenDevice(0,0,false,0);
CKernelOptions m_Options;

CKernel::CKernel (void)
{
}

CKernel::~CKernel (void)
{
}

boolean CKernel::Initialize (void)
{
	return TRUE;
}

TShutdownMode CKernel::Run (void)
{
	m_Screen.Initialize();
	m_Screen.Write("Test message\n",13);
	
	CString Message;
	
	CGPIOPin PIN2(2, GPIOModeInput);
	CGPIOPin PIN17(17, GPIOModeInput);
	CGPIOPin PIN18(18, GPIOModeInput);
	

	while(1)
	{
		m_ActLED.On();
		CTimer::SimpleMsDelay(1000);

		Message.Format("PIN2 PIN17 PIN18\n");
		m_Screen.Write ((const char *) Message, Message.GetLength ());

		Message.Format ("%d  %d  %d\n",PIN2.Read(), PIN17.Read(), PIN18.Read());
		m_Screen.Write ((const char *) Message, Message.GetLength ());

		m_ActLED.Off ();
	}

	return ShutdownReboot;
}

`

xlar54, 2021-08-10:
raspberry pi 3b+

hpingel, 2021-08-10:
Please check out this sample:
https://github.com/rsta2/circle/blob/master/sample/03-screentext/kernel.cpp

rsta2, 2021-08-10:
You need to instantiate `CDeviceNameService` too, because `CScreenDevice::Initialize()` calls `CDeviceNameService::AddDevice()` to register the screen device in the name service. Please try:

```
#include "kernel.h"
#include <circle/gpiopin.h>
#include <circle/timer.h>
#include <circle/devicenameservice.h>
#include <circle/screen.h>
#include <circle/koptions.h>

CDeviceNameService m_DeviceNameService;
CScreenDevice m_Screen = CScreenDevice(0,0,false,0);
...
```

xlar54, 2021-08-13:
got it, ty!

closed 2021-08-13 by xlar54