rsta2/circle #244: HDMI display not working properly.

stay26, 2021-08-11:
Board used : Raspberry pi 2
I am trying to run sample code on my raspberry pi. But there is blank bar all around the display. I had tried many display but same result.
Thanks in advance.
[image]

dwhinham, 2021-08-11:
Try `disable_overscan=1` in `config.txt`.

stay26, 2021-08-11:
Thanks for reply but i don't know where is config.txt file.

stay26, 2021-08-11:
Thanks, issue is solved.

stay26, 2021-08-12:
On my monitor bezels at top and bottom still there. Only side bezels are gone.

rsta2, 2021-08-12:
Circle has no influence on the overscan display settings, this is done independently by the firmware. You can only try to eliminate the borders with the *config.txt* settings in [this section](https://www.raspberrypi.org/documentation/computers/config_txt.html#generic-display-options), but not sure, if this will work for all display resolutions.

stay26, 2021-08-12:
Thank you for fast reply. Now i understand the whole concept of display drivers.

stay26, 2021-08-12:
Thank you for fast reply. Issue is solved.

closed 2021-08-12 by stay26