rsta2/circle #245: SPI1 on pi-zero not working

benmcevoy, 2021-08-12:
Using the 23-spisimple example on a Raspberry Pi Zero.

If i enable USE_SPI_MASTER_AUX and wire in a loopback on the pi-zero between pins GPIO20 and GP19 as described in the README I see the correct data being logged out.  Which is correct and expected.

However, when I use a real device - a MCP3008 ADC on SPI1 CE0 I get no response.  

The same code using SPI0 CE0 works without issue.

I send three bytes

	TxData[0] = 1;
	TxData[1] = 0xB0;
	TxData[2] = 0;

which instructs the MCP3008 to read the value on channel 3 of the ADC, and gets no response (all zeroes) back.

I do not have an oscilloscope so I cannot see what is going out on the wire :(

I also tried sending the bit stream "in reverse", e.g. { 0. 13. 128 }, to test if LSB first made a difference and no luck there. 

I tried another SPI device - Pimoroni Unicorn HD hat, again, works fine on SPI0 but swapping to SPI1 and it does not work.  The only change is to wire into SPI1 and swap CSPIMaster references with CSPIMasterAUX.

rsta2, 2021-08-12:
The signals on a RPi Zero with your modifications to *sample/23-spisimple* using SPI0 look like this:

[image]

Using SPI1 (AUX) they look like this:

[image]

Despite a small delay before raising `/CS` again, this is quite the same.

Unfortunately I do not have a MCP3008 here for testing, but I will order one, so that I can check this. This will take some time.

rsta2, 2021-08-24:
I got the MCP3008 today. Unfortunately I cannot reproduce your problem with SPI1 (AUX). Using a RPi Zero with *sample/23-spisimple* with these modifications:

* Enable #define USE_SPI_MASTER_AUX
* #define TEST_DATA_LENGTH 3
* Setting before m_SPIMaster.WriteRead():

```
TxData[0] = 1;
TxData[1] = 0xB0;
TxData[2] = 0;
```

The test program dumps the following bytes with the given input voltage on channel 3:

```
0V	00 00 00
1.65V	00 02 00
2.2V	00 02 AA
3.3V	00 03 FF
```

I think that's right. I connected the MCP3008 as follows:

```
MCP3008			RPI GPIO

4	CH3		varying input voltage
9	DGND		GND
10	/CS SHDN	18
11	DIN		20
12	DOUT		19
13	CLK		21
14	AGND		GND
15	VREF		3.3V
16	VDD		3.3V
```

benmcevoy, 2021-08-29:
Thanks for looking into this.

I am still having this issue, building from the master branch da58255
No troubles on SPI0 with chip select 0 or 1, all is fine.
SPI1 on CE0 gives no results.
I use the exact same wiring, boot into raspberry pi os instead and do get the expected results.

I'm tending to believe I am doing something wrong, as I am not very experienced in C or electronics, really just started in the last year :) so if you want to close this issue as "can not reproduce" then no worries from me :)

I will try the develop branch and the addon/sensor.
Meanwhile I solved my actual problem with the project i was working on and just put both SPI devices on SPI0 and used two CE lines.

rsta2, 2021-08-30:
Thanks for responding. I think, I will keep this issue open for the case, that somebody has a similar problem.

rsta2, 2023-04-08:
No similar problem so far. I think, this can be closed.

closed 2023-04-08 by rsta2