rsta2/circle #246: QEMU keyboard issues

incanus, 2021-08-13:
I've gotten a project working fine on Pi hardware, but in QEMU (on both macOS and Linux hosts) I can't get `08-usbkeyboard` or other USB examples to work, regardless of various tweaks to QEMU startup arguments related to USB. It's worth noting that even without any explicit USB configuration, the QEMU `info usb` shows the generic QEMU mouse and keyboard. However, the Circle samples all die with log line `dwhci: Unknown vendor 0x4F54294A`. 

If it matters, I'm on an M1 Mac, and my (Debian) Linux "host" testing is actually with a Parallels VM on said Mac, but USB seems to otherwise working properly there. 

Any ideas on how to debug this?

rsta2, 2021-08-13:
Recent QEMU versions require [this patch](https://github.com/smuehlst/qemu/commit/62b39b245b1909c370a365aa2e93fd44830240d2) to work with Circle USB.

incanus, 2021-08-13:
Thanks — looks simple enough (other than building QEMU from source). I had naively tried changing that vendor ID line but wasn't sure of the implications. It also wasn't clear to me that the `smuehlst/qemu` fork was still necessary. I see now that I missed it in the `qemu.txt` doc — my apologies.

closed 2021-08-13 by incanus

incanus, 2021-08-13:
For anyone else having issues on M1, [this gist](https://gist.github.com/nrjdalal/e70249bb5d2e9d844cc203fd11f74c55) is incredibly useful. I just manually applied the USB patch afterwards. Otherwise, I could build the fork on Linux, but not macOS.