rsta2/circle #247: Splash screen change

stay26, 2021-08-14:
Is there any way to change splash screen logo with other image.

rsta2, 2021-08-14:
Circle itself does not have a splash screen. If you mean the rainbow screen, displayed by the firmware, you can switch it off with the [disable_splash](https://www.raspberrypi.org/documentation/computers/config_txt.html#disable_splash) option in *config.txt*.

stay26, 2021-08-16:
I want to change that rainbow screen with other image. I already disable it in config file.

rsta2, 2021-08-16:
There is no special support for splash screens in Circle, but you can draw one on your own using the `SetPixel()` method of the class `CScreenDevice`. Have a look at *sample/02-screenpixel* for that. Another possibility is to use the class `C2DGraphics`, which is demonstrated in *sample/41-screenanimations*.

stay26, 2021-08-16:
Thanks, i will go with `SetPixel()` method.

closed 2021-08-16 by stay26