rsta2/circle #250: CM4 emmc

jamesbond67, 2021-09-11:
How to use compute module with emmc.

rsta2, 2021-09-12:
It seems that the bootloader is looking for the file *recovery4.elf* and doesn't find it. Unfortunately I cannot tell you, why the bootloader is doing this. Normally the file *recovery4.elf* is used to update the bootloader itself, but not for normal operation. Do you have a kernel image *kernel7l.img* and the files *start4.elf* and *fixup4.dat* written to the eMMC?

This problem is not caused by Circle. You should setup a Linux system first on your CM4 using [this doc](https://www.raspberrypi.org/documentation/computers/compute-module.html#flashing-the-compute-module-emmc) and when it is working, you can replace the kernel image *kernel7l.img* with an image, which was built from a Circle sample program.

jamesbond67, 2021-09-13:
Yes i have written kernel7l.img, start4.elf and fixup4.dat to emmc. But no luck.

jamesbond67, 2021-09-13:
I have solved the issue by first booting cm4 to raspberry pi lite os and than replacing kernel7l.ing with generated kernel7.img.
Thank for this grate library and support.

closed 2021-09-13 by jamesbond67

jamesbond67, 2021-09-13:
Now everything works fine but if i use usb in code cm4 gives error.
`logger: Circle 44.1 started on Compute Module 4 (AArch64)
00:00:00.76 pcie: Link down
00:00:00.76 pcie: Cannot initialize PCIe bridge (-1)
00:00:00.76 xhci: Cannot initialize PCIe host bridge`
So i define `#define USE_XHCI_INTERNAL`  in sysconfig.h file but usb is not working.

HenryJunW, 2021-09-27:
How do you replace the kernel7l.img with the generated one? Since I am not able to modify any files in the /boot folder. Thanks!

rsta2, 2021-09-28:
There is a separate tool available with which the eMMC partitions can be accessed from a host computer. Please read [this](https://www.raspberrypi.org/documentation/computers/compute-module.html#flashing-the-compute-module-emmc) for more info.