rsta2/circle #251: CM4 serial communication issue

jamesbond67, 2021-09-17:
CM4 works fine while transmitting serial communication data but it doesn't work while receiving data. My serial initialization is as below.
`m_Serial(&m_Interrupt, TRUE) `
`m_Serial.Initialize(115200);`
Serial bootloader also doesn't works.

rsta2, 2021-09-17:
I tested *sample/32-i2cshell* from Circle 44.1, which uses the same serial initialization, again on a CM4 and it's working well. Also the Flashy bootloader is working. You have to enable the system option `USE_XHCI_INTERNAL` in *include/circle/sysconfig.h* so that this sample program works on the CM4.

Please ensure to have the following files on the eMMC:

* armstub7-rpi4.bin
* bcm2711-rpi-4-b.dtb
* bcm2711-rpi-cm4.dtb
* config.txt
* fixup4.dat
* kernel7l.img
* start4.elf

These files will be provided in Circle's *boot/* subdirectory, when you enter there:

```
make
make bootloader
make armstub
```

This is described in *boot/README*. *config.txt* is the renamed *boot/config32.txt* file. Please note that there may be problems with very recent firmware files, due to a incompatible change in the firmware, so it is recommended to use the files downloaded in *boot/*.

Please try the bootloader with a reduced baudrate, because 2 Mbps may be too fast. I'm using `FLASHBAUD = 921600` here. This setting must be done before `make bootloader`, because the baudrate is hard-coded in the bootloader image.

jamesbond67, 2021-09-18:
I have not added `bcm2711-rpi-4-b.dtb`. I thought it was for rasp-4 but after adding it everything works fine.

closed 2021-09-18 by jamesbond67