rsta2/circle #253: About fatfs

jamesbond67, 2021-09-25:
In [fatfs](https://github.com/rsta2/circle/tree/master/addon/fatfs) how to access second partition on emmc.

rsta2, 2021-09-25:
The multi partition feature of FatFs is disabled by default in Circle (`FF_MULTI_PARTITION = 0`). If you want to access multiple partitions on a volume, you have to activate it in [ffconf.h](https://github.com/rsta2/circle/blob/master/addon/fatfs/ffconf.h#L184) and provide a `VolToPart[]` table, which defines the partition mapping. This is described [here](http://elm-chan.org/fsw/ff/doc/filename.html#vol). Personally I have not tried this so far.

jamesbond67, 2021-09-25:
I will try it it and let you know if it works.
Thanks

jamesbond67, 2021-09-25:
Thanks for the help. By defining `#define DRIVE		"1:"` i can write into second partition.

closed 2021-09-25 by jamesbond67