rsta2/circle #258: Raspberry Pi Zero 2 W support

rsta2, 2021-10-30:
The *develop* branch has been updated to be used with the new Raspberry Pi Zero 2 W. Because it uses the SoC, which was introduced with the Raspberry Pi 3, you have to specify `RASPPI = 3` here. Both 32-bit and 64-bit modes should work. Unfortunately the WLAN device is not working yet.

You should use the latest firmware for the Raspberry Pi Zero 2 W, which can be downloaded in the boot/ subdirectory after removing the comment character from [this line](https://github.com/rsta2/circle/blob/master/boot/Makefile#L9).

dwhinham, 2021-10-30:
Thanks Rene!

I've sent PR #259 to take care of Wi-Fi.

rsta2, 2021-10-31:
Thanks to Dale, Wi-Fi is working now too. I'm currently not aware of anything, which is not working on the RPi Zero 2 W.

dwhinham, 2021-10-31:
Excellent! :slightly_smiling_face:

I've just realized something that we probably need to take care of.

It seems that the Pi Zero 2 W does not respect the `[pi3]` conditional model filter in `config.txt`, therefore we will need to amend `config64.txt` so that a 64-bit kernel can be booted successfully on this model.

In my brief testing, `[pi02w]` and `[pi0]` seem to work, for example, this causes a 64-bit kernel to boot on my Zero 2 W:
```ini
[pi02w]
arm_64bit=1
kernel=kernel8.img
```
However, I am not sure if `[pi02w]` is correct, as it was just a guess. I am not sure how strict the string-matching is, e.g. if `[pi02w]` may unintentionally apply to the original Pi Zero; I don't currently have one available for testing.

Currently, this does not seem to be documented in the official table yet, so I have opened this issue against the RPi documentation repo: https://github.com/raspberrypi/documentation/issues/2254

rsta2, 2021-10-31:
Good point! I think, we should wait for the documentation issue and then update *config64.txt*. In *config32.txt* there is currently no special setting for the RPi Zero 2 W needed. Thanks.

rsta2, 2021-11-01:
The RPi documentation has been updated. The official filter for the RPi Zero 2 (W) is `[pi02]`. I added such entry to the file *boot/config64.txt* on the *develop* branch. Thanks again Dale for helping to figure this out.

dwhinham, 2021-11-01:
Perfect, thanks again Rene!

rsta2, 2021-12-02:
The Raspberry Pi Zero 2 W support has been released with Circle 44.3, so this can be closed. Thanks again @dwhinham for helping to get this going.

closed 2021-12-02 by rsta2