rsta2/circle #262: Cannot Get Sign of life from UART (was working on previous tools version)

tonoso, 2021-11-09:
I just updated to the latest code from a previous baseline based on the old tools (gcc-arm-9.2-2019.12-mingw-w64-i686-aarch64-none-elf). I am finding that I get no UART comms when I move to the latest code and new toolchain (so no sign of life..., only some LED pulses at the start). I have tested both 32bit and 64 bit architectures on RPI3 and RPI4. I am setting the cmdline.txt as before (logdev=ttyS1 fast=true) but get not UART comms. Note that I am using 115200 bps. Please let me know of any changes that I must make to get the code running, or shall I revert to a specific point in the repo? Any help is appreciated. Thanks.

rsta2, 2021-11-09:
I built the *sample/21-webserver* from Circle 44.2 AArch64 on Windows 10 again using the environment described in *doc/windows-build.txt* (the first variant). I used the 10.2.1 toolchain from the ARM website noted in the *README.md* file. I had a problem with the latest 10.3.1 toolchain. Ich got an error message about a DLL, when *aarch64-none-elf-ar* gets called, which can be ignored. 

The kernel images built with both toolchains did run successfully on a RPi 4B. The log output was sent successfully via the serial interface (`logdev=ttyS1`). I wasn't able to test this on Windows itself, because I do not have a driver for my serial USB interface there, but it worked with *Putty* on Linux and the kernel image must be all right hence. I personally never used the serial interface on Windows, but at least building a working program should successfully work.

If your problem remains, I need more information about your development setup, the (sample?) program you trying to run, the Circle release (version), what happens (does the LED blink 5 times?) and maybe your serial adapter. Also which firmware version do you use? Have you installed the *armstub8-rpi4.bin* and *config64.txt* (renamed to *config.txt*)?

tonoso, 2021-11-10:
Hello, Thanks for your support on this. I have tried a few things now and managed to make some progress:
- The LED blinks 5 times
- I can see the log on an external HDMI display when I change to logdev=tty1

Note that I also installed the previous tools (10.2.1), but the same issue remains with the UART connection, I cannot see a log on Putty. I tested both sample 29 and sample 21, as you pointed out. I have confidence that the USB serial adaptor works fine, as it does work with a previous version of the code (around 1 year ago old).  Note that I am using the aarh64 version of the tools, but I did also test 32 bit and got the same results (no UART comms). So details as follows:
- Sample 21 or 29
- Circle release 44.2 and 44 (tested on both)
- LED Blinks 5 times, log shows up correctly on HDMI (logdev=tty1) but not on UART (logdev=ttyS1)
- UART is attached to UART0 pins (GPIO 14, 15). This was proven on a previous program
- UART is configured as 115200, 8 data bits, 1 stop bit, no parity, XON/XOFF flow control (also tried no flow control)
- The serial adapter is a Prolific based adapter.
- I use the FW from 'make' on root and /boot of these releases
- I have installed the files pointed out and renamed to config.txt

Let me know if you have any ideas of things I can try to get this UART working. In the meanwhile, I'll use the external monitor as the debug log. Again, thanks for your support.

rsta2, 2021-11-10:
Attached there is a test program for the UART in the .zip archive. I included the source code and a ready built and tested kernel image for the RPi 4 in AArch64 mode. Please copy the kernel image onto your SD card and start your RPi with a HDMI screen and a terminal program attached to the UART (GPIO pins 14 and 15 according to [this](https://pinout.xyz) at 115200 Bps 8-N-1).

The program sends a message to the screen and the UART every second and displays data, which it receives via the UART, on the screen too. So you should see "Hello screen!" every second on the screen and "Hello UART!" every second on the terminal program. I tested this here and it works.

So if you do not get the output on the terminal program, there must be something wrong with the terminal program or serial adapter (e.g. wrong COM port selected) or the UART TX pin of your RPi is dead. You can also connect the GPIO pins 14 and 15 together for a loopback. In this case the "Hello UART!" message is displayed on the screen too. You can test this way, if your RPi UART is OK.

[uart-test.zip](https://github.com/rsta2/circle/files/7510789/uart-test.zip)

tonoso, 2021-11-10:
Hello, Thanks for all of this.
I have tested the attached .img and also my code. I observe that comms on UART only work when I have a display attached to the board. I see the logging on the terminal and the display when this is the case. However, if the display is removed, no logging is observed on the Putty serial terminal.
I conclude that, somehow, there is a dependency between UART logging working and having a display connected. Note that I also observe this with my older code. On RPI3, it logs to UART without display attached, but on RPI4, it only logs to UART if the display is attached. This makes sense to me now. Let me know if you want me to try anything else out. Does this theory make sense? do you see such dependency in the code somewhere? 
Thanks and best regards! 
Antonio

dwhinham, 2021-11-10:
> On RPI3, it logs to UART without display attached, but on RPI4, it only logs to UART if the display is attached.

This is correct, because of: https://github.com/rsta2/circle/blob/master/CHANGELOG.md#L266

You can either remove the initialization for CScreenDevice or define `SCREEN_HEADLESS` as described here:

https://github.com/rsta2/circle/blob/master/CHANGELOG.md#L243

tonoso, 2021-11-10:
Amazing! I'll give that a try. Cheers very much.

rsta2, 2021-11-11:
I wouldn't be able to explain this better than @dwhinham did it. Does it work now for you @tonoso?

tonoso, 2021-11-11:
Yes, the solution proposed by @dwhinham  works fine. Consider this closed. I would emphasize this change somehow and also update cmdline.txt, which states:
				("tty1" for screen (default), "ttyS1" for UART,
				 "ttyS1" should be used if no screen is attached))

I think it could state that on RPI4, a screen must be attached, or a change to make it headless is needed (#define SCREEN_HEADLESS ... ). Just a suggestion.

rsta2, 2021-11-12:
I will add a note to *doc/issues.txt*, that the RPi 4 requires `SCREEN_HEADLESS` to run most of the samples without screen attached.

rsta2, 2021-11-22:
I think, this can be closed.

closed 2021-11-22 by rsta2