rsta2/circle #263: Could MCP300x support provide an API for the "raw" value from the adc?

diyelectromusic, 2021-11-09:
Unless I'm missing something, I believe the MCP300x support appears to only return the absolute measured (scaled) voltage from the analog inputs.

Could a method be added to return the raw digital value being read?  This mirrors what many might expect to happen if they are used to programming microcontrollers (for example from analogRead with Arduino), where you expect the analog to digital conversion to return the digital value from the conversion not a scaled analog value.

Many thanks for circle, this is amazing work!

rsta2, 2021-11-09:
Yes, of course this can be added. I will try to provide a version for test until weekend. Thanks for appreciating Circle!

rsta2, 2021-11-10:
The updated version is on the *develop* branch.

diyelectromusic, 2021-11-11:
Look's perfect!  Many thanks :)

rsta2, 2021-12-02:
Reading the raw values is now supported in the Circle release 44.3, so this can be closed.

closed 2021-12-02 by rsta2