rsta2/circle #264: Unable to debug on pi 3

niraj-vividmotion, 2021-11-15:
Using serial remote. PL2303 adapter connected to GPIO 14 and 15. 

I copied prebuilt kernel7.img from turboscrew/rpi_stub, set up cmdline.txt as mentioned in instructions. Set up USE_RPI_STUB_AT define and recompiled circle and sample 03.

On booting pi, gdb was launched and following commands were executed:

```
(gdb) set serial baud 115200
(gdb) target remote /dev/ttyUSB0
Remote debugging using /dev/ttyUSB0
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
Remote replied unexpectedly to 'vMustReplyEmpty': timeout
(gdb) file kernel7.elf 
Reading symbols from kernel7.elf...
(gdb) load
You can't do that when your target is `exec'
(gdb) load kernel7.elf 
You can't do that when your target is `exec'
```
```
arm-none-eabi-gdb --version
10.3-2021.07 (arm-10.29)
```

Am I missing any step?

rsta2, 2021-11-15:
I tried rpi_stub on the RPi 3B and unfortunately it does not work any more. There have been multiple changes to the RPi firmware since 2016, when rpi_stub was last updated and it's likely, that it is not compatible any more. I'm thinking about removing the special rpi_stub support from Circle.

But there is an alternative debug stub for GDB: [Alpha](https://github.com/farjump/Alpha_Raspberry_Pi_Circle). I tried it again and it worked with Circle 44.2 with the firmware, which can be downloaded in boot/. Please read the instructions in the Alpha repo.

Because of changes to Circle in the meantime, you have to apply this patch to the *load.gdb* file in the Alpha repo:

```
diff --git a/load.gdb b/load.gdb
index 78c9a18..59aa966 100644
--- a/load.gdb
+++ b/load.gdb
@@ -70,7 +70,7 @@ set $Alpha_serial_write = $Alpha_address + 0x1500
 # Add execution to the Alpha memory space in MMU Page Table
 tbreak CMemorySystem::EnableMMU()
 command
-        set m_pPageTable0Default->m_pTable[ $Alpha_address / 0x100000 ].XNBit = 0
+        set m_pPageTable->m_pTable[ $Alpha_address / 0x100000 ] &= ~0x10
         continue
 end
 
@@ -90,7 +90,7 @@ command
      continue
 end
 
-tbreak *CSerialDevice::CSerialDevice(CInterruptSystem*, bool) 
+tbreak *CSerialDevice::CSerialDevice(CInterruptSystem*, bool, unsigned int)
 command
        p/x $cpsr
       HOOK_GDB_CALL_FUNCTION_AT_END_WHEN_STOPPED_BEFORE_PROLOG $Alpha_serial_pin 0xF00
```

niraj-vividmotion, 2021-11-16:
Thanks. I was able to get Alpha working using this patch.

rsta2, 2021-11-18:
rpi_stub still works! I tested it again and found out, that one has to add the option `rpi_stub_enable_neon` to the file *cmdline.txt*. It's also recommended to explicitly set the serial baud rate for rpi_stub and GDB. I have updated the info for using rpi_stub with Circle on the *develop* branch.

rsta2, 2021-11-22:
I think, this can be closed.

closed 2021-11-22 by rsta2